opentrack/opentrack #1398: UDP 6DOF from a remote machine?

PaulNNaish, 2021-12-29:
Hi,
How do I read 6DOF from a remote machine (on which I wish to impliment the head tracking algorithms)?
I set opentrack input to UDP over network, click the hammer and all I see is a port number and yaw, pitch and roll.  I'd expect to see X,Y & Z entries and a IPv4 address or hostname field
I want to get this resolved before writing the socket server code.

Regards Paul

sthalik, 2021-12-29:
Read double-precision IEEE floats in the order of X, Y, Z, yaw, pitch, roll. The size of each datagram is even `6 * sizeof(double)`.

PaulNNaish, 2021-12-29:
but how does it find the server? does it do a search of the entire network using the specified port?

PaulNNaish, 2021-12-29:
[image]

PaulNNaish, 2021-12-29:
This is of particular relavence but it is "the wrong way round": it is reading the output from opentrack
https://github.com/opentrack/opentrack/issues/1203

PaulNNaish, 2021-12-29:
[image]
I'd expect to see this dialogue box on the input settings rather than output settings

PaulNNaish, 2021-12-29:
The Python code lifted from 1203 gives an error

_#https://github.com/opentrack/opentrack/issues/1203
import socket, struct
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
#sock.bind(("192.168.0.119", 4242))
sock.bind(("127.0.0.1", 4242))
data = sock.recvfrom(48)[0]
for i, label in enumerate(["X:", "Y:", "Z:", "Yaw:", "Pitch:", "Roll:"]):
    print(label, (struct.unpack('d', data[i*8:(i+1)*8])[0]))_

gives the following error message 
_OSError: [WinError 10013] An attempt was made to access a socket in a way forbidden by its access permissions_

sthalik, 2021-12-29:
@PaulNNaish,
> I would expect to see the server's IPv4 address to the left of the port number

It listens on `INADDR_ANY`.

If you can't see the data being sent, then use `tcpdump` or Wireshark. On Windows, packet sniffers can't see loopback traffic. So try sending to some unused IP address, public or private.

sthalik, 2021-12-29:
Also, you can verify the address it's listening on via `netstat` on Windows or Linux, or Process Explorer from the sysinternals suite.

PaulNNaish, 2021-12-29:
This is getting beyond me. I wonder if you could please help me create a suitable Python script to serve the 6 bytes to opentrack please, rather like https://github.com/Cecemel20 has done in
https://github.com/opentrack/opentrack/issues/1203?

sthalik, 2021-12-29:
```python
#!/usr/bin/env python

import socket, struct

address = ("127.0.0.1", 4242)
buf = bytearray(8 * 6)
data = [1, 2, 3, 4, 5, 6]

with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as sock:
    while True:
        struct.pack_into('dddddd', buf, 0, *data)
        sock.sendto(buf, address)
```

PaulNNaish, 2021-12-29:
That's great thank you. I now have it working between two machines.  It's really counter intuitive that the server uses the clients address, I would have thought that only the client would need an address for the server, rather like a webserver.

I'll need three links for each of the three computers (left, centre & right). I'll report back.

PaulNNaish, 2021-12-29:
All working now thanks.  Just got to work on control flow now (blocking/non-blocking behaviour).

sthalik, 2021-12-29:
The server is the one who receives. The client is the one who sends. It has to be like that in a one-side-only, stateless protocol.

Blocking is mostly irrelevant, but the data you send is only sampled 250 times a second. So there's no need to send more than that, except perhaps as a way to control jitter for very precise devices.

What are you implementing anyway?

PaulNNaish, 2021-12-30:
I'm running X-Plane 10 with three screens from three networked computers, which is great because you get the necessary peripheral vision so you can find runways etc.     I can't justify going VR because of the atronomic prices of high end graphics cards (GTX3080 is about £2500 I believe). I'm not keen on TrackIR etc because rotating one's head while keeping your eyes forward seems unnatural to me, especially as I have three screens.

I believe a much better solution is to to track your head and to translate and zoom the displays to give the correct parallax so that the monitor bezels appear to be the windscreen surrounds.  You don't need much parallax correction to fool the brain into believing the scene is "out there". Without paralax correction the image appears absolutely fixed to the monitors. If you have the space for projectors the detrimental effect of this parallax error is much reduced and I'm tempted that way but it's a lot to put out and put away if you don't have a dedicated room.

I was going to have an array of four cheap webcameras viewing my computer desk and observe them with a Raspberry Pi running a Python script (using the opencv library) to calculate in space the position of a single LED mounted in the centre of a pair of glasses.  If you think about it you will realise that you don't need roll information as the scenary will be locked in the roll axis to the monitors and so you only need a single LED, four cameras, opencv and a lot of trignometry. I was going to get the Pi to do all this and feed the three instances of x-plane the necessary zoom, pan and 'yaw' numbers. If you lean into the screen you should almost be able to look back through the side windows. 

Notice this is in the past tense because I went away last night energised with what I thought was a solution only to discover that x-plane zoon (FOV) is locked to the mouse scroll wheel and cannot be assigned to a "joystick" axis (an opentrack DOF in other words).  Until I can resolve that I'm stuck because parallax tracking requires that.  I'll be wading through Google 'Hits' for the next few day but I may have to give up.

Shame because I reckon that parallax tracking would be a very convincing and immersive alternate to VR for flight simulators.

Best regards and thank you 
Paul

GO63-samara, 2021-12-30:
This is a very interesting solution.
But maybe tracking and calculating trigonometry from three cameras will be too difficult?
Will this reduce the FPS in the game?
Maybe it makes more sense to use inertial tracking?
Look here 5DOF tracker (in Russian):
https://sites.google.com/site/diyheadtracking/устройства-обзора-для-авиасимуляторов/razor-ahrs-5dof-20191019-самодельный-инерционный-5dof-трекер
Good luck!

PaulNNaish, 2021-12-30:
The computational overhead of calculating the trigonometry is unknown to me and that is why I intend(ed) to use a fourth computer to do this and calculate the inputs for opentrack on the 4th computer and simply pass the calculated camera positions to opentrack instances running on the simulator machines using UDP. Actually the problem isn't as bad as it sounds because you only have a single LED and triangulating on that shouldn't be too difficult (arctan function).

It's so frustrating to be so near and yet so far away. I'll have a look at other simulators and see how they function as it's unlikely that I (we?) will be able to hack x-plane and get around the mouse-scroll-zoom problem.

I might copy the entire thread and email it to the authors of x-plane.

PaulNNaish, 2021-12-30:
Of course it't the focal length of the camera you need to associate with the forward backward DOF in opentrack, not the actual position.

PaulNNaish, 2021-12-30:
And you will need roll because of vanishing point issues in the side windows but we need focal length control of the 'camera' first.