opentrack/opentrack #1400: Is it possible to adjust gain and exposure for ps3 eye open drive?

AlphabetGhost, 2021-12-31:


AlphabetGhost, 2022-01-01:
Hi i'm using ps3eye camera with point track, for open drive i can't adjust gain and exposure value so ir tracker does not work at all as cl driver, is there a way to set the gain and exposure value?

GO63-samara, 2022-01-01:
Yes it is possible.

`PointTracker Settings` - `Camera` - `Camera settings (when available)` - `Open`:

[image]

Don't forget to click `OK`

AlphabetGhost, 2022-01-02:
> Yes it is possible.
> 
> `PointTracker Settings` - `Camera` - `Camera settings (when available)` - `Open`:
> 
> [image]
> 
> Don't forget to click `OK`

grateful!

closed 2022-01-11 by sthalik