opentrack/opentrack #1405: Cap pose is flipping / lost after side points overlap or leave camera's view

PawelEsc, 2022-01-07:
Hello,
the problem I am trying to fix is that the view is getting flipped sometimes after cap's side diodes are overlapping for a short while. This started with the problem of flipped output after the diodes leave the camera view, but I think it might be relate to this issue.

I tried to fix that by using the Dynamic pose setting, which as per https://github.com/opentrack/opentrack/issues/289#issuecomment-171885479 "helps remember which point is which." This example shows that it doesn't always work. When the points overlap, even for duration shorter than the pose timeout, there's a chance that the pose will get flipped. I display this in a brief recording:
https://gfycat.com/thunderousmajoraustralianfreshwatercrocodile
I hope I understood the purpose of dynamic pose correctly, and the yellow cross can be used to find out that the pose has switched.
While I see that after they overlap it might not be possible to know which is which, but it appears that the model is rotated (90 degrees of roll), so not even the top-center diode's identity is followed, as then only yaw should be affected. Could this mean, that after failure to resolve side diodes the model is "reinitialized" and the top-center diode is incorrectly chosen again using some logic (e.g. brightest one)?

---
The other issues with LEDs leaving camera's view are shown here, although I understand that these might be impossible to be fixed:
https://gfycat.com/talkativecookedconch

An idea from technical side of view is that there could be some user-set constraints added, e.g. up to 90 degrees of input yaw to each side, as it's not possible for the LEDs to be visible to the camera when user is facing away, so the fit of the model should never allow for the left-side diode on the raw input to be interpreted as the right-side diode of the model.

Hope that's understandable and would be glad to provide more data if needed.
- Paweł

sthalik, 2022-01-08:
The points are getting flipped on your video because dynamic pose resets on detection failures.

See:  https://github.com/opentrack/opentrack/blob/master/tracker-pt/point_tracker.cpp#L126

The problem with treating caps geometrically is that the center point can go higher than previous ones on pitching down, and go further to the side when yawing. Now, when pitching and yawing (and maybe rolling a bit too) at the same time, I fail to see how geometry could be used to fix this.

PawelEsc, 2022-01-10:
I though about this a bit and I understand, that the problem is with the identification of the points after all 3 are visible again. I got three rough ideas:

1. Don't discard the pose completely but keep track of identity of the diodes that are visible and unambiguous. I'd say, that as long as the centre diode is known, there's no reason for it to be misidentified when (one of or both) side diodes go out of view.
2. Assume that the user is roughly facing the camera, just as it works when starting the tracking for the first time. This gets ambiguous above 30-45 degrees of roll for typical caps with the centre diode on top of the head, and will break for caps which have diodes visible beyond "overlapping yaw", but will work when returning from pitching down so much that sides go out of view, e.g. looking at side panels in flight sim.
3. As the last orientation known from all 3 points can be stored, after they are back in view the algorithm could go through all options (6 for 3 points) and try to find out which one, after fitting model to the input, produces the closest orientation to the stored one. This hinges on the user "returning" to the viewport "the same way" they quit it. Will fail in the overlap example if the diodes are still seen after around 80 degrees of yaw for typical vantage point. In my case they aren't.

Well, these are not really geometrical solutions, more like brute-forcing, and come with pitfalls themselves. Maybe could work as more robust in some cases, but less flexible in general, optional setting?

FYI the cap I'm using is the Allegro one, "FreeTrack Cap" that you may know of.

sthalik, 2022-01-11:
You can combine (2) and (3) for a less brittle approach. Poses with yaw, pitch or roll over +-60, or Z below 40, can be immediately rejected. Set the reprojection error for these arrangements to a thousand or something.