opentrack/opentrack #1406: Major Jitters After the First Use After Install

ANArmstrong, 2022-01-08:
I've been using OpenTrack successfully for years, but recently I started getting major jitters where the camera rotates by many degrees for one frame, then jumps back to where it was while also having many smaller movements that aren't normally there. If I install OpenTrack again then these problems go away for the first time I run the program, but if I close it and reopen it then the jitters come back right away. The jitters do not appear in the input numbers, they only appear in the output numbers and I have changed all relevant settings without being able to improve the output. The only thing that works is reinstalling the software, and that only works for one use.  The effect is highly reproducible so let me know if you need more data.

sthalik, 2022-01-09:
> The only thing that works is reinstalling the software, and that only works for one use.

This is impossible. The only thing opentrack stores aside .ini files is the profile name in registry.

Are you using the latest version? If so, see if the same happens on 2021.1.2.

ANArmstrong, 2022-01-09:
Thank you for the quick response! It turns out I was clicking a shortcut to an older version so the only reason it worked after the install is because the installer automatically opened the new version and subsequent times the shortcut I use took me to an old version. Thank you again! For completeness: versions 2021.1.2 and 2.3.13 had jitters, but the latest version, 2022.1.0 works great.

sthalik, 2022-01-09:
Glad that it works. Thanks for confirmation.

closed 2022-01-09 by sthalik