opentrack/opentrack #1408: DCS open beta not detecting open track output

aguLepe, 2022-01-12:
I update opentrack today to test the neural net new feature. Opened opentrack, then opened DCS, and it worked smooth, I was exited doing calibration and moving curves to create a new profile. After a reboot, I did the same, opened opentrack, then opened DCS but DCS is not detecting the opentrack output as an input. I mean, in opentrack the octopus is moving and trcking my face, but nothing happens in DCS. I tested it with Elite Dangerous, and it works without problem. Have anyone any idea on how to work around this?

nuxil, 2022-01-22:
Sounds like a issue with DSC. 
But you are vauge with your settings.

Under optput. Do you have freetrack 2.0 Enhandec selected ?
If so, in its setting. What interface is selected?  

It says there, disable one of the protocoles if the game is confused.
You could try disabeling one of them. 

Alltho i am playing DCS myself and i have selected Enable Both. and it works greate

aguLepe, 2022-02-07:
I could be able to fix this. Opentrack profile was corrupted. Thanks for your response @nuxil

closed 2022-02-07 by aguLepe