opentrack/opentrack #1414: How can I reduce latency?

arthurkehrwald, 2022-01-23:
I'm building an installation using a transparent LCD to project content onto real objects. I'm using an infrared clip with the point tracker and I send the tracking data to Unity via UDP. The webcam is a PS3 eye camera running at 75 fps and I display my application on a fast 144hz display. The latency is not bad, but since I'm mapping stuff to real objects, it's still noticeable. I'm wondering if there is any way to reduce it. I played around with filter settings already. I am open to using other hardware, or different tracking / transfer methods. Any suggestions?

vikingsail, 2022-03-03:
afaik a few things affect the "perceived" latency. I say perceived because there is the true latency (delay between the game and OpenTrack, delay between the tracking device and your computer when using wireless, etc) and "added in" latency (caused by filtering, smoothing, etc).

So the first part is relatively small by perception in comparison. The only thing to consider is not to use a wireless connection between the tracking device and OpenTrack (for example some face-recognition based smartphone apps).

The second part is in my experience the most crucial part. I find for example the Hamilton filter is a lot stronger and adds more latency comparing to Accela. Also with "smoothing" set to high will also decrease the responsiveness.

Gorrex, 2022-05-05:
Hey im trying to reduce it too, my findings is that 2 things affect latenct, 
1 Point size, lower point size make the movement more responsive but add jitter
2 filtering, 

I dont know why point size affect responsiveness and stability but it does, if the points your camera get are too big and you cant make it 2 px or less it will add latency. you can try to redo your build trying to make the points as small as possible, but it will have some jitter.