opentrack/opentrack #1415: Two cameras with same name, only one is available to OpenTrack

hrafnkelle, 2022-01-24:
I have two webcams connected to my Win10 PC. Both show up in device manager as "UVC Camera" (both are based on raspberry pi with the ovc-gadget tool). Only one of the cameras shows up in the PointTracker list where I can select cameras.

I've tried changing the "friendly name" of one of the cameras through regedit and that name shows in the device manager now. Still only one camera is available in the Point Tracker camera selector dropdown list.

It seems to be from reading the code that
a) cameras are identified by their name so if one has two cameras with the same name, only one is available
b) its not the device friendly name that is used but the name of a "filter"(?)  that is used.

Are there any unique identifiers other than name that could be used that would enable me to select the right camera?

nuxil, 2022-01-25:
I dont know the code for how its getting the cameras.
Are you are willing to modify the source and do a custom build?
Since youre on windows you could always use windows dshow to get info about your cameras.

dshow will give you things as.
* Description,  "Not cameras all will report back this"
* FriendlyName. 
* DevicePath.

The device path for your two identical cameras must be different. so you could use it.
WIN32 API Example: https://paste.debian.net/1228295/
How to implement this into opentrack i woudnt know.

sthalik, 2022-01-25:
Are you sure that the `modules/` directory inside your opentrack's installation doesn't contain duplicates of `opentrack-video-opencv.dll`? You could have them named like `opentrack-video-opencv - Copy.dll` or similar.

nuxil, 2022-01-25:
There must be more to it than just the camera-names.cpp
I been quickly skimming at some of the files in video-opencv directory.

[impl-metadata.cpp](https://github.com/opentrack/opentrack/blob/master/video-opencv/impl-metadata.cpp#L9) caught my eye. looking at line 9 for starters. 

 int idx = camera_name_to_index(name);
 how would this work if two cameras with same name ?? 
 
 Using names  should be avoided imo. there must be a better signature to use to distinguish between cameras.
 Device path would be uniqe. but not sure how it would work for other platforms.

sthalik, 2022-01-25:
> how would this work if two cameras with same name ??
> Using names should be avoided imo. there must be a better signature to use to distinguish between cameras.

It would pick one in an arbitrary order.

You could append the GUID to the camera names as returned in `compat/camera-names.[ch]pp`.

sthalik, 2022-01-25:
Go here: https://github.com/opentrack/opentrack/blob/master/compat/camera-names.cpp#L78

Copy-paste the variant handling from FriendlyName to also do the same for Device Path. Then just do `str += ' '; str += QString{var2.bstrVal, std::wcslen(var.bstrVal)};`.

Then opentrack should save the unique camera name.

See if it works. If it does then I'm going to add a slightly cleaner version.

hrafnkelle, 2022-01-25:
This works and will also show a capture stream without a physical device (e.g. OBS virtual camera)
 ```
                hr = pPropBag->Read(L"FriendlyName", &var, nullptr);
                if (SUCCEEDED(hr))
                {
                    // Display the name in your UI somehow.
                    QString str((QChar*)var.bstrVal, int(std::wcslen(var.bstrVal)));

                    hr = pPropBag->Read(L"DevicePath", &var, nullptr);
                    if(SUCCEEDED(hr)) {
                        str += ' '; 
                        str += QString{(QChar*)var.bstrVal, int(std::wcslen(var.bstrVal))};
                    }

                    ret.push_back({ str, (int)ret.size() });
```

But the camera names are very long :)
[image]

sthalik, 2022-01-25:
Technically, you could run `wcsrchr(3)` on characters `'#'` and `'\\'` verify that none are `NULL`, and that the pointer of the latter is larger than the former. Then append the substring (which should be the GUID) in lieu of the full device path.

If you can make a clean patch (factoring out variant read into a static function without copy-pasting, it should be `const QChar*`), then send it as a pull request and you're going to be credited as a contributor in release notes.

All that remains beyond that is writing a migration for existing users but I can do that.

hrafnkelle, 2022-01-25:
Let me see what I can do.. thank you

hrafnkelle, 2022-01-25:
The two DevicePath look like this
`
\\\\?\\usb#vid_1d6b&pid_0104&mi_00#8&1699a243&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\\global
\\\\?\\usb#vid_1d6b&pid_0104&mi_00#7&3dff19e&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\\global
`

After removing what is common, then this is remaining:
`
8&1699a243
7&3dff19e
`

I havent been able to find a specification for the DevicePath string format.
Do you suggest picking out the unique part above?

sthalik, 2022-01-25:
Use [wcschr(3)](https://en.cppreference.com/w/cpp/string/wide/wcschr) on the `#` and `\\` characters. Then compare pointers like I said whether the data is valid just to be sure. It's just like `strchr(3)`.

The start of the GUID should be at the pointer to the hash character `+ 1`, and the total length should be pointer to backslash `-` pointer to hash - 1.

hrafnkelle, 2022-01-25:
This logic extracts a guid which is the same in both strings. I think it is not a device guid but a driver guid and both devices have the same driver.
{65e8773d-8f56-11d0-a3b9-00a0c9223196}
So the guid will not help create unique names.

I found this link that explains the format of the DevicePath
https://community.silabs.com/s/article/windows-usb-device-path?language=en_US

Might have to be a little more complex pattern matching to extract the unique part. I'll look into it.

nuxil, 2022-01-25:
How about something rather simpler instead?
Instead of messing about tying to parse the devicepath. 
Take each devicepath and create a small hash for it. ie crc32, md5

Also. i think all this needs be redone in the source.
Where we have the: name of the camera. index of the camera and a hash to distinguish between them.
And in the gui dialog they should be listed with names and index. the hash should not be shown.

sthalik, 2022-01-25:
The PT tracker's dialog should embed inside the options window since the last release (at least it should!). So a GUID won't stretch the window needlessly. Using a separate display and internal name for the camera would make the code more complicated.

sthalik, 2022-01-25:
Alright. So use this instead: <<https://doc.qt.io/qt-5/qregularexpression.html>>.

sthalik, 2022-01-26:
[issue-1415-camera-names.patch.txt](https://github.com/opentrack/opentrack/files/7938317/issue-1415-camera-names.patch.txt)

See if this works for you.

hrafnkelle, 2022-01-26:
This works thank you.
I also added 3 more different usb cameras to my system just in case they got some different formatted device paths. All seemed to work and I could switch between them correctly.
[image]

sthalik, 2022-01-26:
Good to know. Cheers.

closed 2022-01-26 by sthalik

Slion, 2022-04-26:
I'm guessing this broke Kinect V2 IR camera.
  See #1461.