opentrack/opentrack #1420: Shuttering problem after Windows 11 inovation

DonArosa1, 2022-01-29:
Please help me. I have problem with shuttering and no smooth movement in MSFS 2020 after Windows 11 inovation. I use opentrack with aitrack for my MSFS 2020. In W10 all run without problem but in new Windows 11 this softwares run very bad. Its unusable for me. FPS in simulation is the same as in W10 about 45-50. But when I start tracking, movement view has very big lag. I think that is something with cpu but I do not know what is it. I try switch off game mode and increase priority for this aplications but nothing help me. I try reinstall but without succes. Thanks.

sthalik, 2022-01-30:
Try setting CPU affinity for MSFS to all except 1 core.

DonArosa1, 2022-01-30:
I try it but without succes. I buy facetracknoir that compare this two software. Facetracknoir run me very similar. Not smothly. I have some problem with cpu usage in windows, simconnect or drivers.

sthalik, 2022-01-30:
What's the priority for MSFS while it's running? Is the original affinity value getting restored on its own? Are opentrack or aitrack processes marked as suspended in the Task Manager while the game is running?

DonArosa1, 2022-01-31:
- Problem is solved: -

I feel embarrassed, but it looks like in the end it was enough to just remove and reinstall Visual C ++ Redistributable x64, which is on the AItrack page. It's running smoothly now. Thank you Stanislaw for your willingness to help.

sthalik, 2022-01-31:
aitrack's author should build it with `-MT`. All libraries shipped with opentrack are built like that, and that's why opentrack doesn't need any VC++ redistributables.

closed 2022-01-31 by sthalik