opentrack/opentrack #1421: How to get 60FPS in Logitech C922?

GBry, 2022-01-30:
As titled. Using Logi Capture doesn't help too.. Only 30fps, up to 40fps if using Logi Capture..

sthalik, 2022-01-30:
Try the MJPEG mode.

GBry, 2022-01-30:
@sthalik Where can I find that?

sthalik, 2022-01-30:
[image]

GBry, 2022-01-30:
I don't seem to have that option.. I'm on Windows, if that helps.

sthalik, 2022-01-30:
Sure. But first confirm whether you have the right amount of FPS in the PT tracker. Otherwise adding it to the neuralnet tracker won't do anything.

GBry, 2022-01-30:
Can confirm that in the PT Tracker I can get 50-60 FPS even without Logi Capture. It's fluctuating.. but it's much better than the 40fps I reported earlier.

sthalik, 2022-01-30:
Alright. I still need confirmation whether it works.

[opentrack-tracker-neuralnet-20220130.zip](https://github.com/opentrack/opentrack/files/7965827/opentrack-tracker-neuralnet-20220130.zip)

GBry, 2022-01-30:
Ticking/Unticking the MJPEG box doesn't produce any noticeable effect in neuralnet.. Can you also put in a resolution textbox? Because this Webcam needs to be set to 1280x720 to produce 60FPS...

sthalik, 2022-01-30:
Neuralnet has a fixed internal resolution. Adding a resolution combobox will make it even more confusing.

GBry, 2022-01-30:
Okay.. Any suggestions on how I might override the camera before it gets inteercepted by Opentrack?

I tried piping it like Webcam -> OBS Virtual Camera -> Opentrack, but it just shows noise. With the MJPEG ticked ON in the neuralnet, it just crashes. So no-go.

Logi Capture method produces sub 40 FPS, but that's it.

Any alternatives, please?

sthalik, 2022-01-30:
It's up to @DaWelter. I don't see any solution that's as clean as those for the PT and Aruco trackers.

JustDesoroxxx, 2022-05-13:
Neuralnet resolution changing would be amazing my Kiyo need to be in 720p for 60 fps

DaWelter, 2022-05-14:
Hm ... I already try to set a really small resolution, assuming that that would selected a nearby resolution that the camera is actually capable of.
```
bool neuralnet_tracker::open_camera()
{
    int fps = enum_to_fps(settings.force_fps);

    QMutexLocker l(&camera_mtx);

    camera = video::make_camera(settings.camera_name);

    if (!camera)
        return false;

    video::impl::camera::info args {};

    args.width = 320;
    args.height = 240;

    if (fps)
        args.fps = fps;

    args.use_mjpeg = settings.use_mjpeg;

    if (!camera->start(args))
    {
        qDebug() << "neuralnet tracker: can't open camera";
        return false;
    }
    return true;
}
```
But I have a shitty webcam which I doubt can do more than 30 fps. So I never tried to improve the tracker in  this regard :-> Idk, @sthalik do you have an idea what could be done? Can we iterate over available resolutions?

sthalik, 2022-05-14:
@DaWelter
> Can we iterate over available resolutions?

List common resolutions and framerates in the UI. When the effective resolution is different than your trained image size, run `cv::resize` on the image each frame.

DaWelter, 2022-05-18:
Okay. There is some merrit to higher resolutions. The default setting was actually so low that the extracted face patch had to be upscaled sometime before it went into the pose estimation network ...

So, how about that: https://github.com/opentrack/opentrack/compare/master...DaWelter:master?

I've also been working on some other stuff like support for recurrent networks which is also in there.

DaWelter, 2022-05-18:
P.S. I'll make a PR sometime soon.

sthalik, 2022-05-18:
You're the only one who understands that code. What you could be doing is locating the user's face initially and using the ROI around it with some wiggle room, then updating it each frame through the data you receive from the neural network. At least this is what the Aruco tracker does.

I'd recommend against relying on using multiple threads. Sims are getting more resource hungry and the amount of CPU cores doesn't increase fast enough.

DaWelter, 2022-05-18:
That's how it works already :-)
But I like the thread setting. Games are using more cores more effectively these days. Otoh if someone rocks a Threadripper with 32 cores or so ... :-D

JustDesoroxxx, 2022-05-18:
> Okay. There is some merrit to higher resolutions. The default setting was actually so low that the extracted face patch had to be upscaled sometimes before it went into the pose estimation network ...
> 
> So, how about that: [master...DaWelter:master](https://github.com/opentrack/opentrack/compare/master...DaWelter:master)?
> 
> I've also been working on some other stuff like support for recurrent networks which is also in there.

Just looked at what you've done and it's amazing, keep up like that, I am so excited to try it out!

n-prat, 2026-02-17:
This is an old thread, but it does come up when searching for c922 pro and `opentrack` so, I can say that to get 60fps working on my setup I need:

- `MJPEG` enabled
- `720p` resolution [this is the *only* resolution that can get 60fps on a C922 pro webcam]
- disable `Low Light Compensation`! [`Camera Settings` -> tab `Camera Control`]
  - be careful with that one b/c it seems it can randomly toggle back on; maybe at night? In any case, get into the habit of double-checking that one if you see the FPS tank 
- and obviously enough threads for the inference to run at 60 fps but that's not really related to webcam settings