opentrack/opentrack #1424: Can't bind socket, the address is unavailable

Relgar, 2022-02-08:
Hi. I'm trying to use Opentrack with AITrack, and when I change it to get input over UDP I get the following problem:

[image]

The default binding is set to 4242 so nothing out of the box has been changed. I'm not sure why this can't bind to the address. I've restarted my PC in case something was holding the port, but I don't see anything is with netstat either.

sthalik, 2022-02-08:
Have you tried changing the port? If it doesn't help, then it might be a firewall or antivirus issue. Is there anything unusual about your configuration or OS choice?

Relgar, 2022-02-08:
I've changed the port - I am inclined to agree it is a firewall issue but apps usually will ask for permission to be let through. I didn't see any request for Opentrack, so I may try manually. I'm using W10 with just Windows Defender / Firewall.

sthalik, 2022-02-09:
Glad to see that you figured it out. opentrack should use SO_REUSEADDR anyway. I'm gonna change it for the next version.

closed 2022-02-09 by sthalik