opentrack/opentrack #1428: Ubuntu: No values for input, output and filter

harsh0596, 2022-02-22:
I'm trying to run opentrack with aruco to play American Truck Simulator (on proton)

I followed the Debian/Ubuntu guide located at https://github.com/opentrack/opentrack/wiki/Building-on-Linux
one thing to note is that the output was placed in `opentrack/` folder instead of `install/` folder as written in the wiki.

after building and then running `./opentrack`, i got an error loading shared libraries:
```./opentrack: error while loading shared libraries: opentrack-user-interface.so: cannot open shared object file: No such file or directory```

I added whichever libraries were needed to `$LD_LIBRARY_PATH` using [this](https://stackoverflow.com/questions/480764/linux-error-while-loading-shared-libraries-cannot-open-shared-object-file-no-s)

After that, opentrack launches but there are no options for `input`, `output` or `filter`
also, when I hit start, i get `DEBUG [/home/<username>/Documents/opentrack/logic/runtime-libraries.cpp:25]: protocol dylib load failure`
[image]

appreciate any help!

closed 2022-02-22 by harsh0596

airimder, 2023-11-02:
my guy really asked the exact question i have and then closed the issue before anyone could document help for the next person

sthalik, 2023-11-02:
Run `make install` and execute from the install prefix.

airimder, 2023-11-03:
Thank you so much! Not sure how I missed that step, but that's exactly what I was somehow missing! Thank you.