opentrack/opentrack #1431: Logitech Streamcam no 60hz on aruco

eexx1, 2022-02-23:
As the title says, the Logitech Streamcam is not doing 60hz on aruco instead it's capped at 30hz, I tried pointracker with the MJPEG compression option ticked and it works (Goes from 60hz dropping down to 50hz then up to 60hz in not even a second) is there any way to fix this issue?
I tried using Logi Capture as input but in-game I experience terrible stutters and latency.
Pointracker performance: 
[image]
Aruco settings and performance:
[image]

sthalik, 2022-02-26:
So here's a build so that you can start using it already:  https://www.dropbox.com/s/dl07kz4sga91iz2/opentrack-master-20220226_01.7z

eexx1, 2022-02-28:
tried it for about 5 minutes and seems to be working perfectly fine, thank you!
should I close the issue?

sthalik, 2022-03-01:
Good to know, thanks for the confirmation. You can close it.

closed 2022-03-01 by eexx1