opentrack/opentrack #1440: UDP over Network: Recieving Packets on open UDP port, but not tracking.

CheersWorld, 2022-03-15:
Hello. 
I am using GimbalTrack to track my head movements. They are then pushed to the network and I can see the packets in the network, with the correct destination port (using Wireshark). 

I have set OpenTrack to use "UDP over network," with the same port as specified in GimbalTrack, which is also the port that I can see as the destination port in Wireshark. OpenTrack however does not recieve any tracking data. 

The port has been opened in the windows firewall. Other firewalls have been disabled. 

I have tried the ports 4747, 4242 and 5555. None of them worked. 
I have additionally tried starting opentrack as an administrator. This has not resolved the issue.

ahmd1987, 2022-05-20:
Did u solve the problem, i had the same issue too🙄

CheersWorld, 2022-05-23:
Yes, I did. I examined the packets sent by GimbalTrack more closely. None of the values changed much at all with my head movement. For me the solution was then to effectively click all the buttons in the app - that is, disabling the rotational axis and reenabling them, changing the sensitivity, stuff like that. At some point OpenTrack started receiving correct data and started tracking properly. Why exactly this happen I unfortunately cannot say.

ahmd1987, 2022-05-23:
try to open forward  port in the router page and it will solve ur problem
(choose both (tcp,udp), ex. port 4242

renevane, 2022-06-04:
> Yes, I did. I examined the packets sent by GimbalTrack more closely. None of the values changed much at all with my head movement. For me the solution was then to effectively click all the buttons [...]

I was having the same problem and your suggestion did the trick. I think it's a problem with the GimbalTrack app.