opentrack/opentrack #1442: Support - Linux: Wine/Proton communication

jonfive, 2022-03-19:
I've got Opentrack tracking, input works great. v2022.1.1-1

I'm trying to figure out what is actually needed for opentrack to communicate via proton. I selected proton, entered the appid and it starts tracking. However, Elite Dangerous' side it doesn't receive input. I can't seem to find steps specifically for proton usage.

Do i need to install opentrack inside of the prefix first? Manually place dll's? 
any insight or links to exactly what needs to take place would be fantastic. Thanks a ton folks!

Priton-CE, 2022-05-25:
I have a similar problem with the game/alpha Star Citizen.
I opened an issue here: #1479 

I created an [opentrack fork](https://github.com/Priton-CE/opentrack-StarCitizen) that fixes that issue here. Maybe it helps until the issues is addressed

Saltkrakan, 2022-07-01:
> I'm not entirely sure this is the way it's _supposed_ to work.
> I ran native opentrack outputting UDP - opened protontricks, browse to ED game folder, dropped the _windows_ version of opentrack into the prefix.
> Used protontricks to run the windows version of opentrack with UDP input and freetrack output.
> Launched the game and tracking works.

I am trying to run Microsoft flight simulator with open track now. After some others methods, I was trying this workaround yesterday. Unfortunately, I was not sucessful to get it work - There is no input in Flight Simulator and there is no input in proton Opentrack via UDP.
I was not playing with UDP settings much, so maybe wrong IP is issue here, but comunication with Flight sim is thing I am mostly interested in. I am not sure, If I run Proton Opentrack in correct wine prefix. Currently, I am not at home, so I cant post here simple bash script I made for launching Proton Opentrack in wine prefix of Flight sim (ID 1250410).

Other way I tryed were running only Linux version of Opentrack with output set to the Proto/wine. I think my problem here is I have custom Steam library path. I think I have similar problem like #1260, But I hadnt enough time yet to try all stuff described there.

sthalik, 2022-10-07:
There's one evil way to fix this. Make a freetrackclient/NPClient version that listens on a local UDP socket. But if this ever gets implemented, no one will care about fixing the Wine protocol since it's more involved. There are clear downsides to this which is why it was never done before.

- overhead of receiving 250 packets per second inside the game's update/render loop (shared memory doesn't require system calls)
- Valve can break stuff again anyway without being accountable to anyone

Another ugly workaround would be to run a Windows version of opentrack from inside the Proton sandbox using the UDP input with no filtering or mapping set up (or better yet, a simple executable that grabs data over UDP and sends it to freetrackclient). A better one would be to use virtual joystick functionality.

Another workaround would be to run the games using non-Proton Wine, but this needs a steamclient """emulator""" or even a """patched""" executable and thus it's too distasteful to officially endorse.

lesliev, 2023-05-07:
@jonfive I did the same as you and it works using a webcam and "neuralnet tracker" as input and "freetrack 2.0 enhanced" as output.

Steps I followed:
1. Install WineTricks and ProtonTricks - I just ran the wget for WineTricks and used Pip to install ProtonTricks.
2. Download the portable 7-zip version of OpenTrack for Windows and extract it
3. Run ProtonTricks, pick Elite, "Default WinePrefix", Browse Files
4. Drop the extracted OpenTrack directory under Program Files (I renamed it from "install" to "OpenTrack")
5. Now close ProtonTricks entirely and run Elite (if you run OpenTrack first, Elite seems to never open)
6. Once Elite is running, run ProtonTricks again and this time pick "Default WinePrefix" and "Run Explorer"
7. With the explorer that opens, find opentrack.exe and double-click it Now you should be up and running.

I tried compiling OpenTrack for Linux but it requires multiple GB of development dependencies and then doesn't have most of the inputs and outputs - and probably won't communicate with Elite anyway. The "Freetrack 2.0 enhanced" output uses shared-memory IPC which works fine with Wine as long as both programs are under the same WinePrefix.

AnthonyHorton, 2024-05-16:
@lesliev Thanks so much for that, I followed the same steps and was finally able to get head tracking working with IL2.

AFAICT, based on reading the OpenTrack wiki and a bunch of Issues, your method (or variants on it) is, currently, the only way to use OpenTrack with Proton.

I also tried compiling OpenTrack for Linux following the instruction on the wiki, but while I could compile it and run it just fine the result was largely useless because it was missing half of the trackers and outputs (no Aruco, no Neuralnet, etc.). 

I imagine that there are ways to get the OpenTrack compiled for Linux with the missing trackers and output, but there's no documentation anywhere that I could find that explains how.

Based on what I've read of @JT8D-17's work it seems that even if I did compile a feature complete version of OpenTrack for Linux it wouldn't work with Proton anyway, because the fact that Proton now runs in containers broke the mechanism that OpenTrack uses to interact with Wine.

It seems like running the Windows version of OpenTrack inside the Proton container with your game is the only option for the foreseeable future.

JT8D-17, 2024-05-16:
> Based on what I've read of @JT8D-17's work it seems that even if I did compile a feature complete version of OpenTrack for Linux it wouldn't work with Proton anyway, because the fact that Proton now [...]

That thing with the script is way old and I've been using the Linux native Opentrack (well, my [fork](https://github.com/JT8D-17/opentrack) actually) very successfully with Proton games for a long time now.
Not with Elite anymore, but any game that supports TrackIR (e.g. DCS).
The trick is to point Opentrack's WINE wrapper to the game prefix and the exact Proton runtime used by the game.

AnthonyHorton, 2024-05-17:
> That thing with the script is way old and I've been using the Linux native Opentrack (well, my [fork](https://github.com/JT8D-17/opentrack) actually) very successfully with Proton games for a long time now. Not with Elite anymore, mind you, but any game that supports TrackIR (e.g. DCS). The trick is to point Opentrack's WINE wrapper to the game prefix and the exact Proton runtime used by the game.

Thanks for the reply, @JT8D-17.  So is it still the case that the Linux native Opentrack can only be used with Proton versions 5.9 and older, or is that out of date information? If still true that seems like a pretty big limitation for use of the native app.

There remains the challenge of getting the Linux native Opentrack built with all the modules I need. I see the install script in your fork, and it certainly seems to be an improvement over the incomplete instructions in the wiki. I'll try using that as my starting point and see if I can work out how to get it to build the Neuralnet tracker.

JT8D-17, 2024-05-17:
> Thanks for the reply, @JT8D-17. So is it still the case that the Linux native Opentrack can only be used with Proton versions 5.9 and older, or is that out of date information? If still true that seems like a pretty big limitation for use of the native app.

No limitation there. I'm using OT mostly with Proton Experimental.

> There remains the challenge of getting the Linux native Opentrack built with all the modules I need. I see the install script in your fork, and it certainly seems to be an improvement over the [...]

On Arch, all I needed to do was to install the  `onnxruntime` package with pacman and after that, rebuild Opentrack. Opentrack's cmake script then automatically included Neuralnet without any extra flags or build script editing.

AnthonyHorton, 2024-05-20:
> No limitation there. I'm using OT mostly with Proton Experimental, which is bleeding edge. If a game has bog standard TrackIR compatibility, chances are it works without much, if any, fuss.

This is excellent news, however in this case you really should update the `README.md` in your fork.  It still says in there, in bold:

> **For this reason, do not try to use opentrack with Proton 5.13 or newer (at the moment) as the wrapper simply will not work.**

That was one of the reasons I believed Linux native Opentrack wasn't going to work for me, and went with running the Windows version inside Proton instead.

> On Arch, all I needed to do was to install the `onnxruntime` package with pacman and after that, rebuild Opentrack. Opentrack's cmake script then automatically included Neuralnet without any extra flags or build script editing.

OK. I have Fedora, and there was an `onnxruntime` package in the repos that I could install. The OpenTrack wiki Linux compilation instructions really need an update that includes all the currently unmentioned dependencies required by the various tracker and output modules.

I tried the build script in your fork and I'm still having some issues building a working version of the app. I already had most of the build requirements installed, but worked out I needed to install `ninja-build` (different package name in Fedora) and `scotch-devel`.  Your install script has a hard dependency on the X-Plane SDK but I don't have X-Plane and assumed this is only required by an output module that I won't need, so commented out the 
`-DSDK_XPLANE="$otsrc/xplane_sdk" \`
lines . At that point it built without errors (just lots of  `QMutex::QMutex(RecursionMode)` deprecation warnings) but the resulting binary is broken. The GUI appears, but the dropdowns for Input, Ouput and Filter are empty and clicking on them, or the setting icons next to them, does nothing.  Not many clues in the terminal, just a `QSocketNotifier: Can only be used with threads started with QThread` when launching and a `WARN Wayland does not support QWindow::requestActivate()` when clicking on the filter settings button.

sthalik, 2024-05-20:
> The GUI appears, but the dropdowns for Input, Ouput and Filter are empty and clicking on them, or the setting icons next to them, does nothing

Run `make install`, then run the installed binary.

AnthonyHorton, 2024-05-21:
> Run `make install`, then run the installed binary.

Thanks, but the build script from @JT8D-17's fork that I'm using includes the install step.

What happened was that I had been confused by the fact that after running the build script there ends up being a `opentrack` binary in _both_ `$otdir/build/opentrack/` and `$otdir/build/install/`.  When I run the former the GUI appears but with no input, output or filter modules as I described above.  When I run the latter, as I should have done, the GUI appears and there are selectable input, output and filter modules.

Unfortunately they still don't include the modules that I need (Neuralnet and Aruco input and Wine output) so it looks like I'm going to have to do some more digging into the build log and `CMakeLists.txt` to work out why those modules aren't being built and/or linked.

JT8D-17, 2024-05-21:
> This is excellent news, however in this case you really should update the `README.md` in your fork.

Done.

Also updated the section in the readme with instructions on how to disable the X-Plane plugin compilation (I hope that this is enough).

> What happened was that I had been confused by the fact that after running the build script there ends up being a `opentrack` binary in _both_ `$otdir/build/opentrack/` and `$otdir/build/install/`. [...]

As per my Readme, the resulting binary is installed into "opentrack-install" by default, which is one level up and right next to the folder the local repository is in. Running the binary from the "bin" folder in there works just fine for me.

JT8D-17, 2024-05-21:
@AnthonyHorton 
On Fedora 40, after installing the required dependencies (see my updated readme or the header of the build script), my fork builds and runs without issues, including neuralnet (`onnxruntime-devel` was probably missing for you).

I've also included the latest X-Plane SDK so that no build script alteration should be required (if you can spare the extra CPU time for compilation and KBytes of storage).

firestack, 2024-09-23:
I've been trying to get this working with Elite. I've tried both "upstream" opentrack and @JT8D-17's fork. 

I've been using [STEAM_COMPAT_LAUNCHER_SERVICE=proton](https://github.com/ValveSoftware/Proton/blob/proton_9.0/docs/DEBUGGING.md#getting-shell-inside-of-the-steam-runtime) to get a shell into the running container of Elite and have make sure that processes within the container can read the `/dev/shm/facetracknoir-wine-shm` file. But so far, the opentrack wrapper process launched by linux opentrack can't seem to inject into the container in which Elite is running. 

If I run the wrapper from within the container via the `steam-runtime-launch-client`
```sh
wine64 /nix/store/<opentrack path>/libexec/opentrack/opentrack-wrapper-wine.exe.so
```
that does connect to Elite and communicates with the linux opentrack client (but fights with the opentrack wrapper that the linux client launches, rapidly switching back and forth between having a connection to the game and having no game id set)

It _seems_ like there should be _some_ way to get a process running within that container (not manually), but I've not figured that out yet and have not found any resources for doing this.

Jess713, 2026-09-13:
Adding a confirmed-working setup for anyone landing here from search (2026, Linux Mint / Ubuntu Noble base, NVIDIA GPU, Steam Proton 11.0, tested with American Truck Simulator).

Like others in this thread, pointing the Linux-native Opentrack's Wine output directly at the Proton binary + prefix (whether via "Steam Play (select appid)" or "System Wine" with a custom path) did not work here — confirmed via `ps aux` that doing so starts a **second, independent `wineserver` process** that doesn't share IPC state with the one Proton already started for the game, even when pointed at the identical prefix path on disk. So nothing written by the bridge is ever visible to the game, regardless of matching the wine binary version.

What actually worked, matching @lesliev's UDP-relay approach upthread:

1. Linux-native Opentrack (built from source, no special flags needed) does the real camera/tracker input, **output = UDP over network** to `127.0.0.1:<port>`.
2. Download a **Windows portable build** of Opentrack from the releases page, and run it *inside* the game's own Proton prefix via `protontricks-launch --appid <id> /path/to/opentrack.exe`. Since this instance is genuinely running inside the same container/prefix as the game (not reaching in from outside), its **output = Freetrack 2.0 Enhanced** actually reaches the game.
3. That Windows copy's **input = UDP over network** (same port as step 1), relaying the Linux side's tracking data through.

Two gotchas that cost real time in case they help someone:

- **A recent Qt6-based Windows release failed to start under Wine** with `Library icuuc.dll ... not found`. Real Windows 10/11 ships ICU as an OS component that a portable app can rely on; Wine's prefix doesn't provide it. An older Qt5-based release (no external ICU dependency) ran fine. If you hit this, try an older release.
- If `protontricks`/`protontricks-launch` fails with `SyntaxError: Invalid file magic number`, that's an older packaged `protontricks` failing to parse a newer `appinfo.vdf` format version. Rather than upgrading the package, set `PROTON_VERSION="<exact Proton folder name>"` as an env var when invoking it — that skips the appinfo-based auto-detection entirely (see `protontricks/steam.py`'s `find_proton_app`), since you already know which Proton version the game uses.

Also: **do not** try to work around a Wine-binary-path bug (if you hit one) by adding symlinks inside Steam's own `steamapps/common/Proton X.Y/` folder — doing so appears to trigger something in Steam's own integrity checking that wiped that Proton install's payload down to a few hundred KB here, breaking every game using it until "Verify integrity of tool files" was run on it (Steam Library → category filter → Tools → right-click the Proton version → Properties).

Launch order matters too, as others noted above: start the game and wait until you're **actually** in the loaded gameplay view (not just a window/menu — first launches can take a while) before starting the bridge process; doing it too early can make the game hang indefinitely.