opentrack/opentrack #1443: Started up OT for the first time in a few months and can't get it to work?

lifeifbrian, 2022-03-20:
I have tried updating to the latest version, apart from that I don't know what to do. If I ever had this problem before I would just restart PC or turn my PS2 IR webcam off/on which would always work. Any ideas?
[image]

JoDaEmPa, 2022-03-21:
You must use a ps3 cam. I don't know if it also works with a ps2 cam!

lifeifbrian, 2022-03-21:
> You must use a ps3 cam. I don't know if it also works with a ps2 cam!

Sry, PS3 eyecam is what I have.

JoDaEmPa, 2022-03-21:
This is the win 10 driver, but it only runs in programs that still work with 32 bit:
https://mega.nz/file/Hs1i3SgK#oxlPeJq_-B5f96Pt-nlYiKgFz6ShHV0amgDZi1djWqs

but maybe this would be an alternative:
https://github.com/opentrack/opentrack/issues/1383

lifeifbrian, 2022-03-21:
> This is the win 10 driver, but it only runs in programs that still work with 32 bit: https://mega.nz/file/Hs1i3SgK#oxlPeJq_-B5f96Pt-nlYiKgFz6ShHV0amgDZi1djWqs
> 
> but maybe this would be an alternative: #1383

The mega file worked after much tinkering in Opentrack, thanks.