opentrack/opentrack #1451: Output yaw,pitch and roll OSC over UDP Support

hockinsk, 2022-04-07:
As in the title. There's lots of music software with OSC yaw,pitch and roll headtracking support, but none of the webcam type headtrackers seem to output the correct OSC.

Menithal, 2022-09-06:
Specifically, the format is VMC OSC. [This is a protocol over OSC](https://protocol.vmc.info/english)

Vtubing Software Like VseeFace or VMC both use the cameras to create a virtual avatar, would be nice to use that same data, passed usually through (VMC OSC) to also control our fpv cameras in games that support TrackIR. This way we do not have any conflicts over web camera usage. For this purposes, it is only the first 3 floats that are useful  for opentrack.

sthalik, 2022-09-07:
Duplicate of #1264. At this time I'm only available for urgent bugfixes. This is going to last for even a few months