opentrack/opentrack #1453: Compilation failed: readproc.h: No such file or directory

huib, 2022-04-10:
I have compiled and used opentrack successfully before, but recently it failed to launch and instead gave the error `error while loading shared libraries: libprocps.so.8: cannot open shared object file: No such file or directory`

Recompiling gave the following error:
```
In file included from /home/huib/opentrack-src/opentrack/gui/process_detector.cpp:11:
/home/huib/opentrack-src/opentrack/compat/process-list.hpp:134:10: fatal error: proc/readproc.h: No such file or directory
  134 | #include <proc/readproc.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

I am on opensuse Tumbleweed, I have installed the packages `procps` and `procps-devel` (see below).
```
# zypper se -s procps
Loading repository data...
Reading installed packages...

S  | Name           | Type    | Version     | Arch   | Repository
---+----------------+---------+-------------+--------+----------------------
   | busybox-procps | package | 1.35.0-23.1 | noarch | Main Repository (OSS)
   | busybox-procps | package | 1.35.0-23.1 | noarch | openSUSE-20210920-0
i  | procps         | package | 4.0.0-1.1   | x86_64 | Main Repository (OSS)
i  | procps         | package | 4.0.0-1.1   | x86_64 | openSUSE-20210920-0
v  | procps         | package | 4.0.0-1.1   | i586   | Main Repository (OSS)
v  | procps         | package | 4.0.0-1.1   | i586   | openSUSE-20210920-0
i+ | procps-devel   | package | 4.0.0-1.1   | x86_64 | Main Repository (OSS)
i+ | procps-devel   | package | 4.0.0-1.1   | x86_64 | openSUSE-20210920-0
v  | procps-devel   | package | 4.0.0-1.1   | i586   | Main Repository (OSS)
v  | procps-devel   | package | 4.0.0-1.1   | i586   | openSUSE-20210920-0
   | procps-lang    | package | 4.0.0-1.1   | noarch | Main Repository (OSS)
   | procps-lang    | package | 4.0.0-1.1   | noarch | openSUSE-20210920-0
```

My understanding of the issue is that opensuse tumbleweed recently updated to procps 4.0.0, and this version doesn't have a `readproc.h`, meaning the latest version of opentrack in master doesn't work with procps 4.0.0. I will attempt to downgrade procps to 3.3.17 for now and see if that works for me.

Tanks for the great work!

huib, 2022-04-10:
It seems there is indeed a problem with the procps package I have (https://bugzilla.opensuse.org/show_bug.cgi?id=1198150), so clearly I'm complaining in the wrong place :) Keep up the good work!

huib, 2022-04-11:
I am reopening the issue as I am made aware that even though there was an issue with procps on my side, it turns out that still in version 4.0.0 the `readproc.h` interface is not available (see: https://gitlab.com/procps-ng/procps/-/commit/d974f064b5a269fe8bd009d1430d17218aaf07b7). So it is my understanding that something needs to happen at opentrack's side to make it work with procps version 4.0.0

sthalik, 2022-04-11:
Try including `<libproc.h>` instead.

huib, 2022-04-11:
When I change line 134 of `compat/process-list.hpp` as suggested, I get the following error when running `make`
```
In file included from /home/huib/opentrack-src/opentrack/gui/process_detector.cpp:11:
/home/huib/opentrack-src/opentrack/compat/process-list.hpp:134:10: fatal error: libproc.h: No such file or directory
  134 | #include <libproc.h>
      |          ^~~~~~~~~~~
compilation terminated.
```
I don't seem to have a `libproc.h` file on my computer and there don't seem to be packages available for me that supply it.

Even if I manage to compile against the header files of version 4.0.0 of procps, I currently can't test it since the issue with the procps package supplying the binaries is still ongoing.

sthalik, 2022-04-12:
`readproctab(3)` was always available from the header that was now deleted. It's not an internal header.

Can you replace `procps-ng` with `procps` then?

huib, 2022-04-13:
I'll be able to try compiling with procps 4.0.0 again later next week. But I don't really understand the suggestion to replace `procps-ng` as I don't seem to have an package with that name even available.

As far as I understand, procps 4.0.0 has been in the works for years, and readproc was removed from the API maybe 5 years ago (see e.g. https://gitlab.com/procps-ng/procps/-/commit/887bb51016253e82ae26fe9d78bc7a5dfbe2ef85), but this change was only released in version 4.0.0 three weeks ago. It looks like there is some documentation on the new API here: https://gitlab.com/procps-ng/procps/-/wikis/newapi

For now I am happily using opentrack with procps 3.3.17, and I feel like maybe opensuse is a bit early with pushing procps 4.0.0

sthalik, 2022-04-14:
Developers of procps-ng broke a very old, fully working API for no reason. My inner optimist says to wait for them to realize their mistake, rather than clean up their mess. There are tens of thousands of projects all calling `readproctab(3)` in the same vein as opentrack. Naturally, they broke all of them.

It seems that despite procps-ng having removed `readproctab(3)` entirely in the commit you linked to, it still exists in the current source tree. Would it serve as an acceptable workaround if you grabbed [`readproc.h`](https://gitlab.com/procps-ng/procps/-/blob/master/proc/readproc.h) from the `procps-ng`  source tree, and copied it over to `/usr/include/proc/` or similar directory where `procps` headers are being installed to?

Switching over to the new API is less than ideal because users can't be expected to use the procps-ng fork on their systems, and maintaining two separate versions for Linux shouldn't be necessary.

huib, 2022-04-14:
> Switching over to the new API is less than ideal because users can't be expected to use the procps-ng fork on their systems, and maintaining two separate versions for Linux shouldn't be necessary.

I agree. It is not weird for old APIs to be updated though. Still, I would definitely expect a decent period of backwards compatibility. I'll see what the idea is behind that. I guess either linux users need to be able to easily have two versions of the library installed in parallel, or procps-ng needs to have a backward compatible API for a few versions. Not an opentrack issue :)

Then this "issue" is becoming more of a (very early) request to, at some point, migrate to the new procps API.

Thanks

profdecoy, 2023-07-02:
For what it's worth with Ubuntu 23.04, `libprocps` and related development files was removed and replaced with `libproc2-0` which is their packaged version of procps-ng 4.0.3.  I was able to install the packages for `libprocps-dev` and `libprocps8` from 22.10 (and probably 22.04 too for now) along side the 23.04 libproc packages and was able to get Opentrack to build.

sthalik, 2023-07-05:
I won't be supporting both versions at the same time. Remind me when all major distros remove `libprocps`, then old support will be removed and new one added.

sthalik, 2023-07-06:
Please test either `master` or e0507257. Thanks.

RichardFevrier, 2023-12-27:
Hi @sthalik, I tried to compile it under fedora 39 with the package `procps-ng-4.0.3-5` provided by the distribution and could not make it (curretly on the tag `latest`).

the-sane, 2025-01-02:
@sthalik `master` is currently failing to build on Arch with `fatal error: proc/readproc.h: No such file or directory` after updating procps-ng to version `4.0.5`.

I ran into this when trying to re-build opentrack because, after the procps-ng update, it fails to launch with `opentrack: error while loading shared libraries: libproc2.so.0: cannot open shared object file: No such file or directory`. I was hoping a re-build would fix that.

Reverting procps-ng to `4.0.4-3` allows opentrack to launch again without the above error.

the-sane, 2025-01-08:
@sthalik Thanks for responding! I'm building in a clean chroot using Arch's [build scripts](https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot#Convenience_way). I've completely deleted the chroot files to make sure it's clean.

The command you asked me to run returns `-lproc2` on my system for both procps-ng 4.0.4-3 and 4.0.5-1

sthalik, 2025-01-08:
A quick workaround would be to add

```cmake
add_definitions(-DOTR_HAS_LIBPROC2)
```

Right above this [block](https://github.com/opentrack/opentrack/blob/master/gui/CMakeLists.txt#L11-L20).

the-sane, 2025-01-08:
@sthalik  Patching that in does solve it, thank you. Do you need anything else from me to help you figure out the fix on your end?

the-sane, 2025-01-08:
@sthalik That helped with the troubleshooting. I tracked it to a poorly written PKGBUILD from the AUR, so there's nothing wrong on your end. Thanks a lot for the help!

sthalik, 2025-01-08:
If it's `opentrack-git`, you could report a bug as a comment on the relevant page.