opentrack/opentrack #1457: opentrack 2022.1.1 crash

impulse200, 2022-04-23:
Hello.
The app crashes when I have game detection on. I my case I would like to start tracking when I launch il2 great battles. Sometimes it works ok. But most of the time opentrack just crashes with following event in windows "Application" logs:
```
Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x61f14c29
Faulting module name: ntdll.dll, version: 10.0.19041.1566, time stamp: 0xbde09443
Exception code: 0xc0000005
Fault offset: 0x0005f3a1
Faulting process id: 0x2a14
Faulting application start time: 0x01d856ed4b3ed7ff
Faulting application path: C:\Program Files (x86)\opentrack\opentrack.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: bcc14f6b-f90d-4fbd-a9a5-a9ab56f3216b
Faulting package full name: 
Faulting package-relative application ID: 
```

This happens when starting the game both with game launcher and direclty with its main Il-2.exe alike. 
The game version is 4.704, the OS is Windows 10 21H2 build 19044.1645 (all available patches installed).

sthalik, 2022-04-23:
When the crash dialog is on the screen, open the Task Manager and create a minidump. Pack it as a zip file and drag it into this comment text box to share it.

It's like this except you want `opentrack.exe` instead of `lsass.exe`:

[image]

impulse200, 2022-04-24:
> When the crash dialog is on the screen

There's no any dialog on the screen. I notice that opentrack is crashed only by no leds on ps3eye camera is lit when the game is starting. And the opentrack.exe process is not there anymore. May be there's a way to make windows generate dump file automatically on process crash?

impulse200, 2022-04-24:
OK. I managed to collect process dump using this [instructions](https://docs.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps). I made complete dump and it came out quite hefty. So I uploaded it to gdrive: https://drive.google.com/file/d/1B-RnHudbHkyrQjSIakg-sSQRjWzUIEPX/view?usp=sharing

sthalik, 2022-04-24:
Says the file is private.

impulse200, 2022-04-24:
> Says the file is private.

Should be available now.

sthalik, 2022-04-26:
Bug in the CL-Eye driver.

[image]

Use the open driver. See wiki for install instructions.

impulse200, 2022-04-26:
Hi. Thanks for taking time to look into it...
The reason I use proprietary driver is that open one shows abysmal performance:
[image]
Tracking is jerky and jittery with such fps from camera.

impulse200, 2022-04-26:
OK. I'll give it a try. Could you please take a look at #1459 as well, please. I afraid that it will take more programming efforts though.

closed 2022-04-26 by impulse200