opentrack/opentrack #1459: rotate virtual head in-place

impulse200, 2022-04-25:
Hello.
I am having issue with opentrack that I cannot rotate virtual head "in place". 
I'll try to explain (I am using IL2 Great Battles with opentrack here): what I would like to do is, f.e. move head forward to the gun sight then turn the head and see back of the plane to check 6. Instead of turning near gunsight, virtual head is going on big "arc" in space resulting me staring right into seat back. Another exaple is that I am unable to check fuel gauges on Yak-1 in IL2 Great Battles since it requires to raise the head, turn to the wing, move close to cockpit wall and look down to the wing root. 
I was able to kind of fixing this by turning on "Disable effect py pitch" check box in "Relative translation" page of opentrack settings. But checking "Disable effect py pitch" stops relative translation from working. I.e. while looking back, leaning to monitor results in virual head move away from back of the plane. And moving sides left/right move head in "wrong" direction.
I have no such issues with trackir5. Could this be implemented in opentrack as well?

sthalik, 2022-04-26:
Why not disable reltrans entirely then? Besides, you can still use the dynamic neck even with it disabled.

impulse200, 2022-04-26:
> Why not disable reltrans entirely then? 
Because it's unnatural shift left to shift camera right in game. 

> Besides, you can still use the dynamic neck even with it disabled.
Not sure what it is. Didn't see any settings like this.

impulse200, 2022-04-26:
May be it's feasible to add an option to shift center of rotation with head movements? Not to break existing using experience, if this is your concern.