opentrack/opentrack #1461: Kinect V2 No IR Sensor

ernest-cpu, 2022-04-25:
I am trying to use easy tracker but the output is video with color. I saw that I have to select Ir sensor from easy tracker in device tab but I can only see Kinect video sensor there. IR sensor and depth sensor work fine though because I can access them in Kinect studio. Does anyone know how to get Ir sensor to show?
[image]

sthalik, 2022-04-26:
Do you have the Kinect SDK installed?

ernest-cpu, 2022-04-26:
> Do you have the Kinect SDK installed?

Yes and everything works in it. I just dont understand how one is supposed to switch to IR Sensor in open track.

Slion, 2022-04-26:
It's broken in 2022.1.1 the Kinect IR camera can't be selected anymore.
I have 2.3.12 installed and it works fine there.
I guess it certainly works in 2.3.13 too cause it seems it's the last release I was involved with.

sthalik, 2022-04-26:
@Slion can you see whether this patch works?

```diff
diff --git tracker-kinect-face/camera_kinect_ir.cpp tracker-kinect-face/camera_kinect_ir.cpp
index e1ef2208..3a33fd14 100644
--- tracker-kinect-face/camera_kinect_ir.cpp
+++ tracker-kinect-face/camera_kinect_ir.cpp
@@ -48,7 +48,12 @@ namespace Kinect {

     std::vector<QString> CamerasProvider::camera_names() const
     {
-        if (camera_name_to_index("Kinect V2 Video Sensor") != -1)
+        auto list = get_camera_names();
+        auto it = std::find_if(list.cbegin(), list.cend(), [](const auto& x) {
+          const auto& [name, idx] = x;
+          return name.startsWith("Kinect V2 Video Sensor [");
+        });
+        if (it != list.cend())
         {
             // We found Kinect V2 Video Sensor therefore we have a kinect V2 connected.
             // Publish our Kinect V2 IR Sensor implementation then.

```

Slion, 2022-04-26:
@sthalik I have not compiled OpenTrack in a while so I can't easily tell you just now…

ernest-cpu, 2022-04-27:
> Alright then confirm whether this build <https://www.dropbox.com/s/o8531xnbm31wy1z/opentrack-test-20220426_01.7z> works.

Hey, Thanks a lot. This worked perfectly!

closed 2022-04-27 by ernest-cpu