opentrack/opentrack #1464: Opentrack's resetting and toggling buttons haven't been working since the Win upgrade

Loobeensky, 2022-05-02:
Hello everybody!

My man has been using the opentrack tracking software with Arma 3 + an Xbox controller for a few months now. He set up some keys to be able to reset and toggle the tracking smoothly while flying his heli. Everything was coolio till he switched the system from Win 7 to Win 10. The tracking itself still works but the keybinding doesn't; the in-game resetting function is kaputt, toggling as well, the game just doesn't respond to it. As soon as he alttabs and goes back to the tracker's window, everything is bueno again. Basically, he can only use the keyboard but for whatever reason not the controller.

Does anybody have any idea how we can fix it? I'm baffled, I've checked everything I could, all fixes to similar problems, nothing seems to be working.

645285985, 2022-05-26:
following

RRRRRm, 2023-05-24:
Same problem.