opentrack/opentrack #1470: Migration Lib Crash - Static Initialization Order Fiasco?

DaWelter, 2022-05-12:
Hey guys,

today i happened to build opentrack on windows again and ran into a crash in this `push_back`: https://github.com/opentrack/opentrack/blob/master/migration/migration.cpp#L85. My Linux build is fine otoh.

I didn't figure out what is happening but I hypothesize the crash was caused by too late initialization of the vector due to https://en.cppreference.com/w/cpp/language/siof.

Anyway, this code looks a little unsafe if I understand it correctly. Better construct the vector on first use.

I'll try it the next days if you're not faster. :-)

DaWelter, 2022-05-13:
Maybe this wasn't the issue. I also get a crash in the code that get the camera names. Not sure what is wrong. Perhaps a broken build?

sthalik, 2022-05-13:
What build environment are you using?

DaWelter, 2022-05-14:
Win32. 
And there is another thing I'm wondering about.  You are still linking to the static runtime DLLs, right? So what if some object changes ownership across DLL boundaries, won't it cause trouble due to different memory allocators of which each DLL would have its own? Are you sure to avoid that? I found a nice discussion and people there recommend the dynamic runtime DLLs: https://stackoverflow.com/questions/8157891/memory-allocation-in-static-vs-dynamic-linking-of-c-runtime.

sthalik, 2022-05-14:
>  You are still linking to the static runtime DLLs, right? So what if some object changes ownership across DLL boundaries, won't it cause trouble due to different memory allocators of which each DLL would have its own? 

The build system doesn't pass `-MT` by default.

I rebuild all source dependencies whenever there's a new compiler version. Otherwise, binary dependencies that are used don't depend on STL ABI.

Another reason for the failure might be a compiler bug. Have you tried upgrading? VC++ 2019 and above worked for me.

DaWelter, 2022-05-14:
Ah, so you build with /MD or /MDd now?

sthalik, 2022-05-14:
I'm building with `-MT` using the toolchain file, but this doesn't apply to your builds.

DaWelter, 2022-05-14:
But then I don't understand how your build can work 100% correctly :-P

And sorry for being inaccurate, I mean asserts being triggered in the memory allocator. For example
[image]

sthalik, 2022-05-14:
> But then I don't understand how your build can work 100% correctly :-P

It can only malfunction if there's a different ABI across object files. I rebuild everything, including Qt. So it never malfunctions.

> asserts being triggered in the memory allocator

Run `-fsanitize=address`.

DaWelter, 2022-05-14:
Furthermore, it seems to be true - memory is allocated in one DLL and freed in another:
[image]

I believe this is only correct with /MD or /MDd. Not with /MT or /MTd. See that stackoverflow thread I linked above.

sthalik, 2022-05-14:
Hold on, so whose binaries are you using?

DaWelter, 2022-05-14:
Always, self-built, win32, with /MT(d) so far. Ofc I would see these asserts only in debug builds.

sthalik, 2022-05-14:
So `-fsanitize=address` should help you.

DaWelter, 2022-05-14:
Thanks, I can try that.

DaWelter, 2022-05-14:
Hm, didn't work exactly as expected ... but my problem seems solved.

With this sanitization option, the linker complained about simultaneous use of the dynamic and static runtime. I presume the sanitization libraries are only pre-built with /MD? Therefore I switched to /MD for Opentrack. Now I have
* win 32 debug build/MDd
* OpenCV also with /MDd as static libs
No more asserts triggered.
Are you sure your /MT is correct? You can even see in my screenshot that the deallocation routine comes from one of the opentrack DLLs. My understanding is that each of the DLLs has its own datastructures for tracking blocks of memory that were handed out with malloc and new. But the DLL's allocators don't know about the others respectively, so freeing memory with another one can cause problems. With /MD there is only one data structure from the runtime DLL because it's shared.
I might be wrong. I find it strange that your debug builds apparently work :-D

sthalik, 2022-05-15:
> Are you sure your /MT is correct? You can even see in my screenshot that the deallocation routine comes from one of the opentrack DLLs. My understanding is that each of the DLLs has its own datastructures for tracking blocks of memory that were handed out with malloc and new.

You can see [`HeapAlloc`](https://docs.microsoft.com/en-us/windows/win32/api/heapapi/nf-heapapi-heapalloc) coming from `kernel32.dll`.

DaWelter, 2022-05-16:
Okay ... but this linked list code from my first screenshot is from the static runtime. Could be only in the debug runtime so it would not cause issues in the release build.

Are you building your debug builds also with `/MT` instead of `/MTd`?