opentrack/opentrack #1475: PS3 Eye Camera not working on Windows 11 21H2

LaurenceGough, 2022-05-17:
Hi all,

I followed the instructions found here:

https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions

I have tried installing the following drivers which say successful (verfied in use in device manager).
libusb-win32 (v1.2.6.0)
libusbK (v3.1.0.0)
WinUSB (v6.1.7600.16385)

but sadly none of these options work with Opentrack.

I have tried using both the 2022.2.0 and 2022.1.0 Opentrack builds.

At first I got the following error when trying to start tracking: PS3 Eye - access denied (insufficient permission)

Now I just get the message "Can't open Camera PS3 Eye driver error. Unknown error." The red LED does come on the camera during this time.

I am on Windows 11, is that the issue? Perhaps using Win 11 version 21H2 is the issue?

I have tried reinserting the camera into different USB 2 and even USB 3 ports direct on the mobo and restarting my PC.

It doesn't appear to need it as it seems to install fine but does driver signing need to be turned off in Windows for any reason?

Any other ideas? I'd like to ideally use the driver which performs well with low latency which is meant to be this open driver from all I've read.

Thank you,

Laurence

LaurenceGough, 2022-05-17:
Update: I fixed it by using an external USB 3 hub. It makes no sense at all, but it worked straight away.
I have a MSI B550 Tomahawk motherboard, so nothing that unusual or old at all.

michaelosthege, 2023-04-27:
@LaurenceGough could you post a screnshot of which USB driver versions you have for all the different devices? (Using the wizard as described in the Wiki..)

I have the feeling that I'm running into a similar issue because of conflicts between libusb drivers of USB/HDMI adapters. See [my comment in #1495](https://github.com/opentrack/opentrack/issues/1495).

idanmiara, 2023-10-13:
Hi,
I used the PS3Eye as my webcamera for Zoom using some old driver. 
It stopped working at some point and I got to this page while googling for a solution.
I was able to follow the instructions and get my PS3Eye recognized and work with opentrack, But it still doesn't get recognized as a camera for Zoom/Google meet. 

Is there a way to use it as a standard webcam?
Otherwise - I have 3 PS3Eye units available as a donation for anyone who uses opentrack and wants to reimburse me for the shipping costs to their address.

sthalik, 2023-10-13:
Try using this repo: https://github.com/AllanCat/PS3EyeDirectShow