opentrack/opentrack #1477: CMake Error at cmake_install.cmake:58 (file):

Attigliuzzo, 2022-05-23:
I'm used to making a deb package for my Debian system but when i try to package opentrack i get:

> -- Install configuration: "RELEASE"
CMake Error at cmake_install.cmake:58 (file):
  file INSTALL destination: /usr/local/opentrack/./share/opentrack/i18n is                                                                                     
  not a directory.                                                                                                                                             
                                                                                                                                                               
                                                                                                                                                               
>make: *** [Makefile:100: install] Errore 1

>****  Installazione fallita. Creazione del pacchetto annullata.

If i launch "make install", it goes smooth. And this is the row 58:

>  file(INSTALL DESTINATION "/usr/local/opentrack/./share/opentrack/i18n" TYPE FILE PERMISSIONS WORLD_READ OWNER_WRITE OWNER_READ GROUP_READ FILES "/media/attila/back_up/flightgear/opentrack/built/nl_NL.qm")

How can i fix this?
Thanks

sthalik, 2022-05-23:
```console
# rm -fv /usr/local/opentrack/./share/opentrack/i18n
```

Attigliuzzo, 2022-05-24:
> ```
> # rm -fv /usr/local/opentrack/./share/opentrack/i18n
> ```

thanks but same problem:
>Install the project...
-- Install configuration: "RELEASE"
CMake Error at cmake_install.cmake:58 (file):
  file INSTALL destination: /usr/local/opentrack/./share/opentrack/i18n is                                                                                     
  not a directory.                                                                                                                                             
                                                                                                                                                                                                                                                                                                                              
>make: *** [Makefile:100: install] Errore 1

>****  Installazione fallita. Creazione del paccheto annullata.

>Pulitura in corso...OK

I think something is changed somewhere, but your code is ok.

sthalik, 2022-05-24:
Paste the output of `make install` after originally removing `/usr/local/opentrack`.

Attigliuzzo, 2022-05-24:
> 

Checkinstall launches the "make install" command and this is the output:

`Installing with make install...

========================= Risultato installazione ===========================
[ 16%] Built target opentrack-compat
[ 24%] Built target opentrack-options
[ 25%] Built target opentrack-api
[ 27%] Built target opentrack-video
[ 28%] Built target i18n-module-video
[ 28%] Built target i18n-module-api
[ 28%] Built target i18n-module-compat
[ 29%] Built target opentrack-csv
[ 29%] Built target i18n-module-csv
[ 32%] Built target opentrack-cv
[ 32%] Built target i18n-module-cv
[ 35%] Built target opentrack-spline
[ 37%] Built target opentrack-filter-accela
[ 37%] Built target i18n-module-filter-accela
[ 38%] Built target opentrack-filter-ewma
[ 38%] Built target i18n-module-filter-ewma
[ 40%] Built target opentrack-filter-hamilton
[ 40%] Built target i18n-module-filter-hamilton
[ 41%] Built target opentrack-filter-kalman
[ 41%] Built target i18n-module-filter-kalman
[ 44%] Built target opentrack-qxt-mini
[ 50%] Built target opentrack-logic
[ 53%] Built target opentrack-migration
[ 55%] Built target opentrack-pose-widget
[ 61%] Built target opentrack-user-interface
[ 61%] Built target i18n-module-user-interface
[ 62%] Built target i18n-module-logic
[ 62%] Built target i18n-module-migration
[ 62%] Built target opentrack-version
[ 64%] Built target opentrack-executable
[ 65%] Built target i18n-module-executable
[ 65%] Built target i18n-module-options
[ 65%] Built target i18n-module-pose-widget
[ 66%] Built target opentrack-proto-fgfs
[ 66%] Built target i18n-module-proto-fgfs
[ 68%] Built target opentrack-proto-libevdev
[ 69%] Built target i18n-module-proto-libevdev
[ 70%] Built target opentrack-proto-udp
[ 70%] Built target i18n-module-proto-udp
[ 70%] Built target i18n-module-qxt-mini
[ 70%] Built target i18n-module-spline
[ 75%] Built target opentrack-tracker-easy
[ 75%] Built target i18n-module-tracker-easy
[ 77%] Built target opentrack-tracker-freepie-udp
[ 78%] Built target i18n-module-tracker-freepie-udp
[ 79%] Built target opentrack-tracker-fusion
[ 79%] Built target i18n-module-tracker-fusion
[ 83%] Built target opentrack-tracker-hatire
[ 83%] Built target i18n-module-tracker-hatire
[ 84%] Built target opentrack-tracker-linux-joystick
[ 84%] Built target i18n-module-tracker-linux-joystick
[ 88%] Built target opentrack-tracker-pt-base
[ 89%] Built target i18n-module-tracker-pt-base
[ 93%] Built target opentrack-tracker-pt
[ 93%] Built target i18n-module-tracker-pt
[ 95%] Built target opentrack-tracker-s2bot
[ 95%] Built target i18n-module-tracker-s2bot
[ 96%] Built target opentrack-tracker-test
[ 96%] Built target i18n-module-tracker-test
[ 97%] Built target opentrack-tracker-udp
[ 97%] Built target i18n-module-tracker-udp
[ 99%] Built target opentrack-video-opencv
[ 99%] Built target i18n-module-video-opencv
[ 99%] Built target i18n-lupdate
[100%] Built target i18n-qm-zh_CN
[100%] Built target i18n-qm-nl_NL
[100%] Built target i18n-qm-ru_RU
[100%] Built target i18n-qm-stub
[100%] Built target i18n-lrelease
[100%] Built target i18n
Install the project...
-- Install configuration: "RELEASE"
CMake Error at cmake_install.cmake:58 (file):
  file INSTALL destination: /usr/local/opentrack/./share/opentrack/i18n is                                                                                     
  not a directory.                                                                                                                                             
                                                                                                                                                               
                                                                                                                                                               
make: *** [Makefile:100: install] Errore 1

****  Installazione fallita. Creazione del paccheto annullata.

Pulitura in corso...OK

Ciao.
`

I removed the /usr/local/opentrack folder, probably it is a issue with checkinstall that creates, at the end a deb package. In the past i was able to generate a package but now i can't.

Attigliuzzo, 2022-09-11:
It was a issue related to checkinstall, used by me to build the package. Now i build the Debian package in another way, without issue.

closed 2022-09-11 by Attigliuzzo

Moini, 2023-02-11:
more info: fstrans=yes is broken in checkinstall. For years... :-(

Attigliuzzo, 2023-02-12:
Thanks @Moini , now i built Debian packages with `dpkg --build`, they aren't compressed but it is better than nothing.

Moini, 2023-02-12:
Thanks for the tip, @Attigliuzzo ! I'll look into that for my own private packages.