opentrack/opentrack #1479: proto-wine creates a registry key with path outside of wine prefix

Priton-CE, 2022-05-25:
While trying to run Opentrack with the game/alpha Star Citizen I found that the proto-wine output creates a registry with a path as value that leads outside the wine prefix.

I belive Star Citizen tries to load that client but fails since the client specified in the path is outside the bottle.

I fixed this problem by disabling the creation of this key in [my fork](https://github.com/Priton-CE/opentrack-StarCitizen/blob/master/proto-wine/opentrack-wrapper-wine-windows.cxx#L70-L71)

Not sure if this is required for other games but it causes issues with Star Citizen and it references a path outside the wine prefix.

sthalik, 2022-10-07:
So you're using the wine path setting to provide multiple versions of the Wine wrapper, each built with a different libwine? This is clearly an improvement, as right now opentrack doesn't have any infrastructure for building multiple wrappers through its build process.

Can you describe the way you build multiple wrappers currently? Ideally we'd allow specifying multiple paths in opentrack's cmake cache and building them all in one go.

Priton-CE, 2022-10-07:
No I am still using the default libwine.
I have very little experience with how freetrack works so I was trying to find anything that looked problematic. The thing I disabled in my fork is the writing of the freetrack client to the registry since that pointed to a location outside the wine bottle/prefix. (Turns out it was only part of the issue. I still have no idea what else the wrapper writes to the registry to mess with freetrack. Running a bottled windows instance cleans up the registry and allows for the usage of the wrapper again. Without running the bottled instance after using the wrapper Star Citizen will simply not pick up any freetrack client.)

However I added the ability to allow the user to use a custom wine version other than the one in the system $PATH variable (since in Lutris you are usually dealing with non system wine). I have not messed with opentracks build configs except for building wine by default.

RichardFevrier, 2022-12-29:
Any idea what could mess up with freetrack @sthalik ?
Running a second opentrack instance inside wine to fix things up does not seem very elegant 🙂
Thanks for the investigation @Priton-CE and of course for the immense work @sthalik 🎉

Priton-CE, 2023-02-25:
I have found the actual issue: Star Citizen required both a freetrack and NPClient path to be specified in the registry. This only happens for people using Both protocols.
My fork now forces path creation for all protocols.

closed 2023-02-25 by Priton-CE