opentrack/opentrack #1480: Track Hat sensor  Software not picking it up after i hit stop

Rothgar57, 2022-05-26:
Hi all, 

New to this forum.  I purchased the new Track hat sensor and Track hat. I am having some issues with the Software. When i open the software, the IR camera is identified and I see the ir dots.  When I hit stop and start again the Program becomes unresponsive for a minute or two.  Then once the software catches up, open track can no longer find the IR camera.  The only way to resolve the issue is to unplug the camera close the program replug the camera and restart the program.  I have the same exact issue with the Track hat version of open track. 

I have been using open track for years with my ED tracker and never had any of these issues.  

The Track Hat sensor ir camera according to the device mgr is running normal.  It is classifed as unspecified usb coms port 4.   

Does anyone have experience with this or similar issue?  Does anyone know how to go about debugging this?

sthalik, 2022-05-26:
Grab this test version from the trackhat branch. There's gonna be a new one soon but this is the most that can be done on short notice.

https://www.dropbox.com/s/0u4b27jtcxc4xtz/opentrack-trackhat-20220523_01.tar.xz

Rothgar57, 2022-05-26:
This new version seems to have fixed the immediate issue.  The new sensor camera is being picked up consistently.  However it takes over 30 seconds for the software to find it.   This is i a big improvement to me thank you for your support.   Hopefully in the future that duration will lessen.