opentrack/opentrack #1481: Tracking Source head plane not in line with actual head plane

Hayden-Fluff, 2022-05-29:
I'm not entirely sure how to really word the title, but my issue stands as such:

Whenever I move the tracker Left or Right relative to my forward, it affects the X and Z axis at the same time, while this is correct with my tracking source (HTC Vive 3.0 tracker) with how I have my base stations mounted, I don't have a way to really do anything about this, I haven't found a method of remapping that lets me map it to how I want.

I feel the original offset is due to how I have my VR space setup, I have my base stations mounted on opposite sides of the room with my forward direction being perpendicular to them (See below for reference)

[image]
[image]

I already tried both SteamVR Room Setup or OVR Advanced Settings to make the foward the forward I'm at while at the desk, but this hasn't done anything to help, I think this is because OpenTrack works with the raw data given by the SteamVR runtime.

This starts to become very apparent in Star Citizen whenever I'm flying a ship, occasionally I'll be way further offset than I'd like because of the mismatch in the mappings, or in MSFS pretty much after any time after a recenter.

I use the Point recentering method as it works best with how I have my tracker oriented (Tracker button facing up with the charging port facing backwards), though any of the other methods yield the same result.

I'm not really sure what a good solution to this would be, maybe something to pre-offset the source data before it's handled by OpenTrack? I'm unsure, but I'd love to hear any advice or other solutions to this issue, as I feel I can't be the only one that has this.

camrox92, 2022-08-19:
Same setup and issue here. Not much info on this situation on the web and I don't think any of the other tracking methods supported by opentrack work similarly.

Hayden-Fluff, 2022-11-11:
A slight update on this, I recently reinstalled OpenTrack on my new system to use with Microsoft Flight Simulator, and I don't experience this issue with the same exact setup, I have absolutely no idea what the difference is between this and my previous setup, but there is some difference. The forward plane does seem to be in line now.

It's a little puzzling, but considering the issue still may exist, I'll leave it open.