opentrack/opentrack #1486: SteamVR not working after recent steamvr update

Frosty704, 2022-05-31:
Ever since SteamVR 1.22, no tracking data is being read by opentrack. (using a vive tracker), I know it's tracking from a test in tundras python tracker test script, but opentrack is not reading any data. Only fix i found was to downgrade to v.15.12 in steamvr betas

Frosty704, 2022-06-02:
> See #1483.

tried, still not tracking

Frosty704, 2022-06-09:
am using a null driver btw, it seems to work if i have my physical headset detected by steamvr

sthalik, 2022-06-09:
Open Windows Console and drag `opentrack.exe` or the shortcut to opentrack onto the window. Press Enter. The app will launch as usual, but also emit debug information into that window. Start tracking and see if there are any errors related to steamvr. For instance, you might get `vr::vrChaperone == NULL` or `steamvr: no valid pose from device` in that window.

So press Alt+Printscreen while having the console window selected to make a screenshot of that window and Ctrl+V it into this text box on Github.

Frosty704, 2022-06-09:
after I did this i got

DEBUG [F:\dev\opentrack\tracker-steamvr\steamvr.cpp:144]: steamvr: no valid pose from device 1 valid false connected true

but then it started working for some reason? thank you for the help, if it happens again and i cant fix it ill make another issue and quote this one

closed 2022-06-09 by Frosty704