opentrack/opentrack #1493: TrackHat sensor tracking issues

FRazor95, 2022-06-09:
Hello everyone,

I just got my new TrackHat sensor about a week ago and until now I wasn't able to get it to work correctly with DCS.
I already reached out to James and he tried to help with some basic setup tips, especially regarding settings in TrackHat Opentrack.
Unfortunately they didn't help and I also tried them myself before, because they are mentioned in all the various Opentrack setup videos you can find on YT. I also used many various profiles from YT and DCS forums, which also didn't really help.

After this long introduction, now on to my actual problem:

Basic tracking of my head movement works pretty well after some tweaking of the settings, but the input the TrackHat sensor captures seems to be very noisy (although I completely darkened my room and Opentrack shows only the 3 dots, also I tried it at night). My camera in DCS constantly moves by a considerable amount both Yaw and Pitch without me moving my head at all. While this can be fixed for center view with using mapping deadzones, it persists if I look slightly off center and then try to focus on something while holding still (for example my gauges in the cockpit, or while locking targets using the helmet mounted sight). Also using filters doesn't fix the problem as I have to set them to max values which in turn completely destroys the precision.

I really tried playing around with any of the available settings and also used TrackHat Opentrack, as well as the "default" Opentrack, having the same experience with both of them.

At this point I believe this must be a problem with either Opentrack or the TrackHat sensor, because the general experience is even worse than using the SmoothTrack app on my smartphone. Unfortunately I currently do not have any tools to try and debug the TrackHat sensor and I do not want to order another PS3 Eye camera for testing purposes just to find out the TrackHat sensor may be faulty or Opentrack having an issue working with the sensor.

Do you guys have any tips on what to try or what the cause of my issue could be? Maybe you are even able to provide me with any debugging tools?

Thanks to all of you in advance!

Florian

MarcinSzczecin, 2022-06-14:
This camera is not present, but try to install version 2.3.13, it has "old diode tracking" as the last one and in this version go to the accel filter and give low values. In this version and below, there is no flashing and the image does not move.

 The next versions of the opentrack have some new LED tracking and there is a slight pulsation, and moving like you described - I have the same on a regular ps3 camera. They can be eliminated with the "point filter" but the use of this filter causes the screen to float and lack of precision. Sorry for my English

FRazor95, 2022-06-14:
@MarcinSzczecin Yes you are totally right and that's why it currently isn't useable for me at all. I guess I'll have to return the sensor and maybe wait for a fix. Thank you for sharing your experience though!

RobertGiesecke, 2022-07-29:
> To be honest I was really disappointed at first because many people hype headtracking and say it is so great, but it clearly wasn't great at all for me.

I also have problems getting trackhat sensor to work as I want it to.

I've been using headtracking for years.

Started with self-built trackers based on accelerometers, then used trackir5 for a long time now.
But I always disliked the trackir software and the fact that they encrypt their came so I can't use it with opentrack...

The PS3 eye thing can work ok, but only after lots of tweaking. (which I did for my sc org mates)

this sensor seemed like a godsend... 200hz and opentrack.

However, it is orders of magnitude removed from the super immersive and smooth tracking I have setup in trackir.
I have trackir with out center deadzone and no curves (everything is linear), and it is super smooth and I completely forget that it is even there.

The trackhat sensor has the weirdest artefacts, like sawtooth movements going while yaw, or micro stutters.
It is never smooth at all.

I really hope that this is due to the current opentrack software. I also ordered the clip for it, bc the owner said it will only work properly with their clip, not with my luke clip.
It is much better with their clip, but still much, much worse than a properly tweaked Ps3 eye camera setup.

sthalik, 2022-07-30:
> this sensor seemed like a godsend... 200hz and opentrack.

This sensor is a 96x96 pixel array.

RobertGiesecke, 2022-07-30:
> > this sensor seemed like a godsend... 200hz and opentrack.
> 
> 
> 
> This sensor is a 96x96 pixel array.

OK, then it was all a waste of time and money.

WTF? How can somebody have such a low expectation for headtracking to think less resolution than trackir 10 years later was a good idea?

thx for bursting that bubble, was really needed

sthalik, 2022-07-30:
> WTF? How can somebody have such a low expectation for headtracking to think less resolution than trackir 10 years later was a good idea?

To be fair, it almost works. I'm still convinced that with a custom cap design with big LEDs but less depth between LEDs (think TrackHat clip vs mini-clip) it can work as well as the PS3 Eye in 640x480.

sthalik, 2022-07-30:
> How can somebody have such a low expectation for headtracking to think less resolution than trackir 10 years later was a good idea?

Despite all that, it can still work well with caps. Clips are more restrictive in that the LEDs overlap a bit too much. It *almost* works.

RobertGiesecke, 2022-07-30:
> > How can somebody have such a low expectation for headtracking to think less resolution than trackir 10 years later was a good idea?
> 
> 
> 
> Despite all that, it can still work well with caps. Clips are more restrictive in that the LEDs overlap a bit too much. It *almost* works.

hmm, I expected a camera with a higher res than trackir5.
It has been many years since that thing was released.
Doing something "good enough" is very cool for tinkerers, but not if you take money for it. Marketing as something great, when it's not even possible to be mediocre.

Sorry, still amazed about that ridiculous 96x96 and I'm also not helpful at all...

FlipBoii, 2022-08-05:
I'm having the same issues, stay away from TrackHat.

Part 1: Using the regular OpenTrack software
https://www.youtube.com/watch?v=t3YjCE-JNsk

Part 2: Using the TrackHat OpenTrack
https://www.youtube.com/watch?v=YD30PcsrPX0

Tried to return it, paid for return shipping. Had it sent back to my door because parcel refused to take it on the plane due to the battery.

I am now out of luck so I decided to try again, it wouldn't find the 3 dots. I checked with my phone and lo and behold the middle IR is malfunctioning. I don't even know what to do at this point....I am out $300 CAD trying to find an alternative to TrackIR5 and still after almost 2 months of ordering I have yet to experience proper head tracking, this is my biggest regret. Could have, should have just bought TIR5. Lesson learned so please avoid.

https://youtube.com/shorts/cefjKPXP3oc?feature=share

JackD007, 2023-12-06:
Got it working. Manually i put the npclient and freetrackclient dll drivers in system32 and their 64 versions in sysWOW64 ....