opentrack/opentrack #1494: Crash using aruco tracking in 2022.2.0

eyelykaynal, 2022-06-09:
Hello,

I began using opentrack today. I have an aruco marker and a PS3 eye camera with the code labs driver in Win10 19043.1706. After tracking starts, opentrack will abrupty exit after a seemingly random amount of time. The icon remains in the system tray but the process disappears. I have tried continuously moving and not moving at all and I cannot seem to find a cause. I've tried different resolutions and framerates. It does not happen using neuralnet. Here is the log from event viewer:

```
Faulting application name: opentrack.exe, version: 0.0.0.0, time stamp: 0x627c061b
Faulting module name: Qt5Core.dll, version: 5.15.3.0, time stamp: 0x62191136
Exception code: 0xc0000005
Fault offset: 0x0002933f
Faulting process ID: 0x25e8
Faulting application start time: 0x01d87be7adff9695
Faulting application path: D:\Program Files (x86)\opentrack\opentrack.exe
Faulting module path: D:\Program Files (x86)\opentrack\Qt5Core.dll
Report ID: 9f3f29d7-5970-4f63-9db3-51fa57319a1b
Faulting package full name: 
Faulting package-relative application ID: 
```

Any help would be appreciated.

Regards,

Eyely

sthalik, 2022-06-10:
Send a crash dump according to these instructions: https://github.com/opentrack/opentrack/issues/1457#issuecomment-1107472663=.

eyelykaynal, 2022-06-11:
There is no crash dialog and the process doesn't remain in task manager but I will generate a dump via the other method later in that issue when I next get a chance.

sthalik, 2022-06-11:
Drag the shortcut or the executable onto a Windows Console and press Return. It's going to launch as normal, but maybe there's going to be something useful on `stderr`.

eyelykaynal, 2022-06-17:
Here's what I get when opening from cmd:

```
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library D:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
[ INFO:0@1.693] global F:\dev\opentrack-depends\opencv\modules\core\src\parallel\registry_parallel.impl.hpp (96) cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
DEBUG [F:\dev\opentrack\compat\camera-names.cpp:117]: failed CLSID_VideoInputDeviceCategory 1
DEBUG [F:\dev\opentrack\csv\csv.cpp:97]: csv: lookup game id 3450
ps3eye: transfer status 4
```

I've tried with CL drivers and open drivers. With CL drivers there is no "failed CLSID_VideoInputDeviceCategory 1" entry but it still crashes with "ps3eye transfer status 4". Sometimes it's instant, sometimes it takes a minute.

Shall I get a dump via [this method](https://docs.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps)?