opentrack/opentrack #1496: need help with freepie tracker on linux

StanberyTrask, 2022-06-10:
howdy all, was trying to setup opentrack and am running into some problems with the freepie tracker on linux.
I can see freepie working through wireshark, trying to send packets to 4242 and i've opened port 4242 through ufw and iptables but there's still no response from the octopus.

the computer i'm running from also has Ethernet, wifi, and is usb tethered to the phone at the same time, i tired all local ip's in freepie with no change.

using ``netstat -tulpn`` while tracking is on i see
```
Proto - Recv-Q - Send-Q - Local Address - Foreign Address - State - PID/Program name
udp6    0        0        :::4242         :::*                      347977/\\?\unix\hom
```
with that being the proper PID for opentrack, so i really have no idea why it isnt seeing anything. i really dont know what else i can try, if anyone has anything to add i could really use the help. thanks

sthalik, 2022-06-11:
I see that it's not listening on udp4. It definitely should. Try this patch.

```patch
diff --git tracker-freepie-udp/ftnoir_tracker_freepie-udp.cpp tracker-freepie-udp/ftnoir_tracker_freepie-udp.cpp
index 3950af0c..12c2d957 100644
--- tracker-freepie-udp/ftnoir_tracker_freepie-udp.cpp
+++ tracker-freepie-udp/ftnoir_tracker_freepie-udp.cpp
@@ -34,7 +34,7 @@ void tracker_freepie::run() {
         Mask = flag_Raw | flag_Orient
     };

-    sock.bind(QHostAddress::Any, (unsigned short) s.port, QUdpSocket::ShareAddress | QUdpSocket::ReuseAddressHint);
+    sock.bind(QHostAddress::AnyIPv4, (unsigned short) s.port, QUdpSocket::ShareAddress | QUdpSocket::ReuseAddressHint);

     while (!isInterruptionRequested())
     {

```

StanberyTrask, 2022-06-11:
ok that seems to have fixed it listening on udp6, its now just listening on "udp"
```
Proto - Recv-Q - Send-Q - Local Address - Foreign Address - State - PID/Program name
udp     0        0        0.0.0.0:4242    0.0.0.0:*                 383535/\\?\unix\hom
```
but still no response from any types of networking that i can throw at it

sthalik, 2022-06-11:
Use the local IP address of the tethering interface. For testing purposes, you can use the regular UDP tracker and send some pose like that:

```console
$ perl -e 'print pack "d6", 1..6' | nc -uw2 127.0.0.1 4242
```

StanberyTrask, 2022-06-11:
yeah the script moves it, with that address and with the tethering address

sthalik, 2022-06-11:
Run tcpdump on the tethering interface and paste a few lines:

```console
# tcpdump -ni interface udp and port 4242
```

StanberyTrask, 2022-06-11:
sorry my pc decided to die for no reason

```
sudo tcpdump -ni usb0 udp and port 4242
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on usb0, link-type EN10MB (Ethernet), capture size 262144 bytes
04:02:16.305162 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.434009 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.461577 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.503958 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.633941 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.661854 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.702328 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
04:02:16.833786 IP 192.168.14.196.47766 > 192.168.14.32.4242: UDP, length 50
```

sthalik, 2022-06-11:
In the freepie app on the phone, see if you're getting readouts for yaw pitch and roll or only the individual sensors. If ypr stays as zeros then the problem doesn't lie with network communication.

StanberyTrask, 2022-06-12:
im getting data from accelerometer, gyro, and magnetometer, orientation is also updating. i thought it might be the app not actually sending the right info but using wireshark i can see that it is in fact sending data that changes over time

[packet dump hex.txt](https://github.com/opentrack/opentrack/files/8885176/packet.dump.hex.txt)
[packet dump raw.txt](https://github.com/opentrack/opentrack/files/8885178/packet.dump.raw.txt)

i can try another version of the app but this version is the one from the newest release

sthalik, 2022-06-12:
Add a bunch of debug printf statements, like this: [freepie-debug.patch.txt](https://github.com/opentrack/opentrack/files/8885310/freepie-debug.patch.txt). Then read the standard error output of opentrack.

StanberyTrask, 2022-06-12:
ok, using the exact diff that you gave it looks like nothing within ``while (sock.hasPendingDatagrams())`` gets run? 
all i am getting is (i removed full path because bloat)
```
./opentrack
qt5ct: using qt5ct plugin
DEBUG [/home/stanbery/Documents/opentrack/opentrack/main-window.cpp:1030]: no preset dir
DEBUG D-Bus system tray: yes
DEBUG[/path:84]freepie: rot 0 0
DEBUG[/path:84]freepie: rot 1 0
DEBUG[/path:84]freepie: rot 2 0
DEBUG[/path:84]freepie: rot 0 0
DEBUG[/path:84]freepie: rot 1 0
DEBUG[/path:84]freepie: rot 2 0
DEBUG[/path:84]freepie: rot 0 0
DEBUG[/path:84]freepie: rot 1 0
DEBUG[/path:84]freepie: rot 2 0
DEBUG[/path:84]freepie: rot 0 0
DEBUG[/path:84]freepie: rot 1 0
DEBUG[/path:84]freepie: rot 2 0
```

sthalik, 2022-06-12:
Sorry, the last line wasn't meant to go outside the read datagram loop. See updated diff
[freepie-debug.patch.txt](https://github.com/opentrack/opentrack/files/8885339/freepie-debug.patch.txt). Now you're only going to get debug info when a datagram is received. That was the original intention.

Other than that, try connecting to some closed port on the phone to see if you're getting a response. `nc -v 192.168.14.196 12345` should get "Connection refused" immediately without it hanging (checked it on phones connected to my local wifi). Similarly, `ping 192.168.14.196` should get a response.

StanberyTrask, 2022-06-12:
ok so. i can ping the phone from my pc, trying to nc a closed port instantly gets refused, and there's no debug messages coming from opentrack while tracking is on and the app is on

sthalik, 2022-06-12:
One more thing, is there anything in `iptables`?

StanberyTrask, 2022-06-12:
ok i guess that did it, something between me flushing iptables and starting up ufw (i *really* need to make it start on launch) got it working, thank you so much for walking me through all this

closed 2022-06-12 by StanberyTrask