opentrack/opentrack #1499: Tracking only working when windowed/ focus issue

Mujanov3737, 2022-06-17:
Hello,

I'm using opentrack 2022.2.0 with a DIY clip, a Wii remote mounted on my monitor, and WiiPointTracker 1.1 as my input. Everything seems to look great as far as opentrack is concerned. The Wii remote connects fine, the sensor is picking up all 3 ir leds, and the game data looks to be translating correctly. However, I'm only able to get the head tracking to work while I have the game in windowed mode.
 
I've also been able to get tracking for a second or so by having opentrack open, moving my head around, then alt-tabbing to the game which correctly moves the view in the game but then the tracking stops after about a second, almost like the tracking stops working since it isn't the window in focus. 

I thought it may be related to a setting in the game, so I tried both Arma 3 and Elite Dangerous and both are having the same issue. No tracking in fullscreen or fullscreen windowed and I can get the tracking to work for a brief moment by having the opentrack window in focus, then alt tabbing to the game. I've installed opentrack to the same drive as the game and I'm running both it and the games as administrator.

Am I missing something obvious? Would appreciate some help.

sthalik, 2022-06-17:
It's a bug. The Wii tracker doesn't have many users so it takes a long time to spot them. Please test the following build <<https://www.dropbox.com/s/c33r8t8ymoavvwz/opentrack-test-20220617_01.7z>>.

In particular:
- it shouldn't crash on tracking stop and should also resume correctly
- tracking should work when the window isn't visible
- there aren't any new issues

If there are no issues, the fix is grounds for a new stable opentrack release, as it wasn't previously known that the Wii tracker was broken.

Mujanov3737, 2022-06-17:
Thanks, Stanislaw. I gave the test build a try and the tracking seems to be working great in full-screen now. I had a little trouble getting the key binds to work in the game but after restarting the tracking and the application a couple times, the key binds were working in-game and centering my screen.

This could just be my inexperience with the software and head-tracking in general. But as for the original issue, it seems to be well resolved in the new test build. I'll continue to test it out more while playing a game in the coming days and I'll let you know if I encounter anything strange.

I thought the original issue may have been related to the "game mode" setting Windows 10 has, as well as my anti-virus having a "gamer mode" that tries to stop unnecessary background applications while in full-screen in a game, but turning those off didn't help alleviate the issue last night when I was trying the latest official release.

Appreciate all the help.

closed 2022-09-30 by sthalik