opentrack/opentrack #1501: [feature request] access to game data coordinates via UDP stream

gulbrillo, 2022-06-20:
Hi there, friendly bunch :)
I am developing a SpaceMouse / XBox Controller / M/KB remote control for opentrack that sends coordinates via UDP (UDP over network input method in opentrack).
The game (Star Citizen) seems to limit x/y/z motion (game data) beyond "50". My remote however does not know when this limit is reached as there is no back channel for game data coordinates.
**A UDP stream of game data coordinates to a specific IP address/port would be a great addition for my use case.** (making the Xbox Controller rumble when you get close to the invisible wall of x/y/z >= 50, and restrict further motion towards that limit so it's easier and more intuitive to back out of the range limit.
Would this be something that's possible to implement? 😬
Thanks a lot for this amazing piece of software!!

sthalik, 2022-06-20:
As a workaround you can read data from the global named memory mapping `FT_SharedMem`. Structure layout is in `contrib/npclient` as `TFreeTrackData`. Don't bother with the mutex parameter defined in that file, it's not being used.

gulbrillo, 2022-06-20:
Interesting. :) I didn't know something like memory-mapped files even existd. Would I be able to read the FT_SharedMem from a 3rd party app running on the same PC?

sthalik, 2022-06-20:
Yes.

gulbrillo, 2022-06-20:
sweet, thank you!! :) I think mmap is the Python library I need for this. Now I'm excited...!! :D This is awesome!

gulbrillo, 2022-06-21:
In case anyone else needs access to the Game Data: I managed to get it to work in Python :)

```python
import mmap
from dataclasses import dataclass
import struct
import time

@dataclass
class FT_SharedMem:
    DataID: int
    CamWidth: int
    CamHeight: int
    Yaw: float
    Pitch: float
    Roll: float
    X: float
    Y: float
    Z: float
    RawYaw: float
    RawPitch: float
    RawRoll: float
    RawX: float
    RawY: float
    RawZ: float
    X1: float
    Y1: float
    X2: float
    Y2: float
    X3: float
    Y3: float
    X4: float
    Y4: float

bytes = 92 # len(TFreeTrackData) == 92

with mmap.mmap(-1, bytes, 'FT_SharedMem') as mmap_obj:

    while True:
        data = struct.unpack('iiiffffffffffffffffffff', mmap_obj)
        TFreeTrackData = FT_SharedMem(*data)
        print(TFreeTrackData.X, TFreeTrackData.Y, TFreeTrackData.Z)
        time.sleep(0.1)
```