opentrack/opentrack #1505: multiple output support

Arbel-arad, 2022-06-26:
add an option to enable more then one output at a time.
i want to use the trackir output and also send udp to another computer at the same time to use a head-aimed turret in a game.

sthalik, 2022-06-27:
Good suggestion and not too hard to implement. But you know that you won't be able to apply separate mappings or filters to each type of the data, right?

Arbel-arad, 2022-06-27:
yes, that's fine. i can do separate mapping on the other computer for the udp input.

NlGHTLORD, 2022-11-01:
I need more than one output.  I need to send one output to freetrack to control my head movement in-game and a second to Vjoy so I can create macros (with Joystick Gremlin) at the same time.  Please keep me in the loop if possible.