opentrack/opentrack #1508: GUI show nothing even blank text

r0achnnan9p, 2022-07-12:
[image]

sthalik, 2022-07-12:
Missing fully installed English fonts perhaps. There's a partial translation for zh_CN but not for zh_TW. Maybe after fixing this issue you feel like contributing?

sthalik, 2022-07-12:
First off, launch from Windows Console by dragging the executable or the shortcut onto it, then press Enter. It should start emitting debug messages.

The default font being used in dialogs is MS Shell Dlg 2.

> How can I fix this problem?

Try this:

- Add or remove programs
- Optional features
- Install all supplemental English fonts

Also this:

- Region settings
- Language
- Add a language (English, United States)

tobascodagama, 2023-11-28:
Since this issue is still marked Open, I'll just add my experience with this same bug. I encountered it after switching to a higher-res monitor and adjusting Windows' UI scaling. I increased this setting ("Make everything bigger") from 100% to 150% to better suit my new monitor. Since then, I experienced the same issue as OP.

The solution is to decrease the "Make everything bigger" setting to 125% or lower _before_ starting OpenTrack. Once OpenTrack is running, this setting can be changed to something higher with no ill effects.

I'm not sure if this is something that can be addressed in OpenTrack, but at the very least I hope this comment might help someone else.

seaeagle1, 2024-06-28:
Did the same thing for me on 125%, so it's something with the scaling, but not only 150%.

sthalik, 2024-06-30:
Any issues with the font `Ms Shell Dlg` perhaps?

closed 2026-09-05 by r0achnnan9p