opentrack/opentrack #1521: Trackhat

FlipBoii, 2022-07-24:
Hello,

This is a repost, I accidentally posted in the Q&A and I think this section fits better.

I am having a hard time setting up trackhat with the "Trackhat Opentrack" and the regular "Opentrack". I play main DCS.

I just got the trackhat clip pro with their new sensor and I am really disappointed. It isn't as smooth as I thought it was compared to YouTube streamers head tracker. I have been trying different ways to set it up and tweak it for the past 3 days, but I am still not getting the same smoothness as I am expecting. I've Googled and there isn't much about trackhat. When I first opened the trackhat opentrack program, the 3 dots were going all over the place and I found that turning the exposure to min value of 16 helped it to seperate the 3 dots, the default value was 39. Even with this minimum value, I still get points merging, more noticeably on the vertical plane. I don't know if this is related to my issue as I've read that bigger dots are more precise. I've tried with low-light, pretty much just my monitor giving off any light source. I tried with my main room lights on, I tried with a lamp and still the same results. The sensor is offset to the right side of my monitor and the tracking clip is attached to the right side of my headphones. I read that limiting the FPS to 60 will make it smooth but still didn't get the results I wanted.

I keep my head as steady as I can but it sways left and right. The pitch axis is a pain to setup and inconsistent when combining with yaw, like when I look over my shoulder and looking up. I find that the points on the vertical axis merge a lot. So I've tried pointing the clip up/middle/down/left/right, off-setting the camera to the right a little bit/a lot. Everytime I want to play DCS, I spend more time positioning the camera/clip and mappings everytime I take my headphones off. Also, I noticed that the "px" size on the 3 dots are constantly changing even when I am still. I tried to place the sensor and clip on my desk so there is no movement but the "px" is jumpy but only by 1 or 2 values. Is this normal ? There's no way I can keep my eyes on a bandit in Air to Air combat with how it is. Also there are times when the trackhat opentrack program will hang (not responding) and will not connect to the Sensor unless I unplug and replug the usb. I've also tried to use the regular opentrack but it seems like its the same since trackhat has their own in the drop down so I can't use point tracker, I got worst results so I stuck with trackhat opentrack.

I really want to like trackhat but its frustrating at this point. My neck really hurts from trying to set it up properly, lol. I don't know if getting a TrackIR 5 will yield the same results if I am not doing something right setting it up. I am desperate and want to know if I should return it and stick with my webcam which I can get close to the same results or get a TrackIR 5.

Recorded 2 videos showing my problem.

Part 1: Using the regular OpenTrack software https://www.youtube.com/watch?v=t3YjCE-JNsk

Part 2: Using the TrackHat OpenTrack https://www.youtube.com/watch?v=YD30PcsrPX0

FlipBoii, 2022-07-25:
> One immediate problem I see is that the max point size is set to 50 px. At 1:06 and 1:09 in the video the points actually go over the 50 px boundary. Try setting the max point size to 100 px or something.
> 
> For keeping the view still, go to the Filter tab and increase Accela rotation smoothing slightly. You can keep the point filter as-is, it's already set to a high value. You might also want to increase the rotation deadzone (on the same Filter tab) but that may give you a different kind of artifacts.
> 
> Tracking quality using the TrackHat sensor degrades heavily once the point size decreases. So starting up with point size of 45 px has the potential of actually working if it's set up correctly. 25 px as it is later in the video, won't work well no matter which filter settings you use.
> 
> I recommend getting a PS3 Eye with the IR filter removed, and a visible light filter already installed. But judging by the video, your TrackHat setup isn't too bad and you might be able to get good results if you spend more time to configure it. That said, there's no version of opentrack that lets you use it without performing any configuration or learning the underlying concepts.

Hello!

Thanks for the reply! I remade a video with the suggestions you made. 

 _"That said, there's no version of opentrack that lets you use it without performing any configuration or learning the underlying concepts."_

I have seen probably mostly all the posts about TrackHat, at least for the new sensor anyway, there's not much out there because the sensor is still new and they were using PS3 Eye prior to the change.   So, I wasn't expecting to use it out of the box, I spent 3 days trying to configure everything from the software settings to the hardware physical placement of sensors and tracking device.

The only reason I mentioned the version is because with the "TrackHat" drop down for the Output in OpenTrack, as opposed to "Point Track", there are lesser options for the camera settings. ie: I don't get to see the width or height or the diagonal field of view camera settings.

I also want to share an email between another competitor.  That suggested that maybe the resolution settings of the camera is causing the issue.  I want to keep the competitor anonymously but I want to commend them for trying to help me with a competitor's product.

[image]

**_"Hi,

If you could give me the model dimensions for your clip and the width and height resolution of the camera in OpenTrack’s camera settings, it would help.

 

My first impression is that the model dimensions are causing the intersect problems and freezing your view.

 

I would have suggested increasing the smoothing filter to help with the drifting, but in the second video, you had it cranked up high with no relief.

 

You could increase the deadzone to hold your head still in the middle, but I don’t think you’d be pleased with the result. You should try it though. 

 

I think the profile mapping curves are too aggressive and making it difficult to hold your view still.

 

Cheers,"_**

Thanks again!

T0meReader, 2022-08-13:
> One immediate problem I see is that the max point size is set to 50 px. At 1:06 and 1:09 in the video the points actually go over the 50 px boundary. Try setting the max point size to 100 px or [...]
How does one increase the point size?  Mine are 17px, 22px, and  25px top to bottom.