opentrack/opentrack #1527: Can't select camera for Easy Track or PointTracker

PrimeEpilogue, 2022-08-14:
Hey,
I have my Kinect for Xbox One (I believe this is the V2, Model 1520?) hooked up to my pc and have installed SDK browser v2.0, all seems to be working in the browser.
When I try to set up OpenTrack to use Easytracker or PointTracker I have no option to select a camera. I'm not sure what I am doing wrong. I have tried the latest release as well as 2022.2.0 and 2.3.13( saw that someone else stated this release was working for them)
[image]

Any ideas on how I can fix or troubleshoot this?
Kinect face 0.1 works but id really like to try out IR tracking

Slion, 2022-08-18:
That's weird, see that [post](https://github.com/opentrack/opentrack/issues/1461#issuecomment-1109950744).

sthalik, 2022-08-18:
That's one old opentrack version.

sthalik, 2022-08-18:
@Slion, the Kinect patch has already been included in 2022.2.0, so it should work just fine in 2022.3.0.

Slion, 2022-08-18:
Just tried 2022.3.0 and it works just fine it seems:
[image]
[image]

sthalik, 2022-08-18:
Try running opentrack from within a console window to get some debug info. To do this easily, drag the executable or the shortcut onto an empty `cmd.exe` window and press Return.

PrimeEpilogue, 2022-08-18:
looks like it is now working, only change I made was I opened it via the CMD window and now it appears to be finding my Kinect even when launching normally.

closed 2022-08-19 by Slion