opentrack/opentrack #1530: Pitch and Roll issue with opentrack 2022.3.0 via aitrack-v0.6.6-alpha.zip through UDP

iNgeon, 2022-08-19:
Just updated to opentrack 2022.3.0 with aitrack-v0.6.6-alpha using UDP
1. When pitching head seems to input more into Y axis than actually into Pitch
2. Rolling head to the right roll goes from 0 to -75 however rolling head to the left translates into Yaw to the left instead.
3. When looking at the AITrack preview the dots seem to track accurately... 
Am i doing something wrong ?

GO63-samara, 2022-08-20:
Maybe wrong Centering method?
See: https://github.com/opentrack/opentrack/issues/1340

dmoshea, 2022-08-21:
I had the same issue. Reverting to aitrack-v0.6.5-alpha resolved it.

AIRLegend, 2022-08-23:
Already fixed on https://github.com/AIRLegend/aitrack/pull/163. Will be with the next version. In the meantime, you can either disable roll or downgrade to 0.6.5.

Sorry for the inconvenience :(