opentrack/opentrack #1532: Y axis recognition doesn't work properly.

lchy0508, 2022-09-14:
I just downloaded Opentrack-2022.3.0 and linked it with the AItrack which is webcam base face recognition program.

The problem is that the Opentrack recognizes the opposite of me moving my face to the y-axis. Which means that when I look up, the octopus looks down, and when I look down, the octopus looks up.

I tried the Y-axis invert option on output settings but it didn't work. And checked the AItrack but it seemed to be working properly.

Does anyone know how to solve this problem?

[image]

GO63-samara, 2022-09-14:
Maybe you need to invert the **Pitch** axis?
The **Y** axis is not rotation, but vertical movement.

lchy0508, 2022-09-14:
> Maybe you need to invert the **Pitch** axis? The **Y** axis is not rotation, but vertical movement.

Thx, I will try that.