opentrack/opentrack #1534: Center Pose hotkeys (VERY HIGHLY REQUESTED!!!)

MrKossmo, 2022-09-24:
PLEASE! Add this feature! It's very useful I should say! And very highly requested by me for few years. I download new versions again and again. And there still no this feature. But this is very useful feature. Why not? I don't think it hard to implement it. There are hotkeys for start tracking/stop tracking, zero center and others primary actions, same as centering pose. But why you do think it's not important to adjust center pose in game while stitting in cockpit, and i should minimize game application few times to adjust correct number for each aircraft every time? Why don't you implement it?

noisytwit, 2022-10-10:
There is already one,..

Options > Shortcuts > Global Shortcuts > Center

I use it all the time in DCS, just move your head where you want the new center, then hit the hotkey and it centers based on your new head position.

MrKossmo, 2022-10-12:
No, this not i'm talking about. I've already bind that action (centering).
I'm talking about functions we have in section

Options > Output > Custom center pose

Yep, moving head and  than press centering button is a kind of "solution"... But it's not comforlable to huddle every time. And it's not so easy to do it precisely and correctly. It becomes obvious when i zoom my view in

It's better it relax, hold head in comfortable neutral postion. And then adjust virtual head zero position with hotkeys without switching software.

Please make it in Opentrack! It's very useful feature! Thank you