opentrack/opentrack #1539: Neuralnet tracker Linux

durian, 2022-09-27:
I checked out and compiled opentrack on Ubuntu 22.04, but I don't seem to be able to get the neuralnet tracker.

Are any special steps needed to get this to work under Linux? I installed the libonnx-dev libs, but maybe more is needed?

DaWelter, 2022-09-28:
There is ONNX and there is ONNX Runtime. You need the latter. You can get it from the releases in the official github repo. Then, configure cmake to point ONNXRuntime_DIR to the directory where you extracted the files.

durian, 2022-09-28:
Thanks! It works now under Ubuntu 22.04.

I also had to add a
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fopenmp")`
to get it work in CMakeLists.txt

CodringherAndFenn, 2025-06-02:
Hello, 3 years later I hope you can still see this. Im also trying to get the neural network tracker to work on opentrack. My problem is that I use Linux Mint (based on Ubuntu 22.04) and the libonnxruntime-dev seems to be part of a very new branch of Ubuntu (25.04 i think?)
How did you manage to get libonnx running on Ubuntu 22.04?
Excuse me if I seem like I lack some basic knowledge, I am quite new to Linux overall

CodringherAndFenn, 2025-06-03:
First of all, thank you for answering to a 3 year old obscure post.

Everything works now, small hiccup after building where it wouldn't find a library even if it did find it during build, but that's solved as well and the Neural Network tracker now works.

Again, thank you so much for this! I can now play my flight sims on Linux!

closed 2025-06-04 by DaWelter