opentrack/opentrack #1541: 2022.3.0 + Win11 + EasyTracker + Kinect IR + FreeTrack + ED = crash

Slion, 2022-09-29:
Everything seems to work fine in Windows but as soon as I switch back to ED OpenTrack crashes. I can see it worked for a brief moment in game before it quits.

Slion, 2022-09-29:
v2.3.12 works

Slion, 2022-09-29:
v2022.2.0 broken

Slion, 2022-09-29:
v2022.1.0 broken

Slion, 2022-09-29:
v2021.2.0 broken

Slion, 2022-09-29:
Can't try v2021.1.3 because Windows detects a virus.
The last version that works with ED for me is 2.3.13.

sthalik, 2022-09-30:
What's the stderr output?

Slion, 2022-09-30:
Looks like a similar issue than: #1499
I'll post the error output later today I guess.
Currently downloading the  latest ED version should could take a few hours.

Slion, 2022-09-30:
Here is relevant output I get:

> [F:\dev\opentrack-depends\opencv\modules\highgui\src\window_w32.cpp:1261] opencv: NULL window: 'Preview' at cvDestroyWindow

Looks like EasyTracker preview window could be somehow causing that issue. I'm going to try run without it.

Slion, 2022-09-30:
Sure enough it works after disabling the EasyTracker debug preview window.
Though even when disabled the window opens up but it is not displaying the feed.
Most certainly an easy fix, not sure when I'll have the time to look into it though.
It's odd that's only an issue when switching to ED.