opentrack/opentrack #1542: Aruco Tracker settings button crashes program

Inhaltsangabe, 2022-10-01:
[image]
If I click this button Opentrack freezes and just doesnt answer and i get the no response message from Windows

uuozz, 2022-10-16:
happening to me too , cant click on any button but start

sthalik, 2022-10-16:
Make me a minidump (with opentrack 2022.3.1):

[image]

uuozz, 2022-10-17:
> 

done it , do i attach it there?
sorry kinda new to githud

sthalik, 2022-10-17:
Pack it inside a .zip, then Ctrl+V into this text box.

uuozz, 2022-10-17:
file is to big :c

sthalik, 2022-10-17:
Did you use Ultra compression?

Or just put it on Dropbox or OneDrive.

uuozz, 2022-10-17:
> Did you use Ultra compression?
> 
> Or just put it on Dropbox or OneDrive.

https://we.tl/t-65mWNCeXF6

sthalik, 2022-10-17:
I updated libusb. See if this helps: https://www.dropbox.com/s/prlbigdqss1hrtp/opentrack-test-20221017_01.7z.

uuozz, 2022-10-17:
btw it happen whit every tracking oprions , nothing seems to change

sthalik, 2022-10-17:
Do you have a PS3 Eye camera connected to the PC?

sthalik, 2022-10-17:
As a workaround, delete `opentrack-video-ps3eye.dll` from the `modules` directory.

uuozz, 2022-10-17:
wow it worked , hope that works to everyone !!
thanks a lot for the patience

closed 2022-11-12 by sthalik