opentrack/opentrack #1554: WiiPointTracker does not appear in Opentrack-2022.3.1

GO63-samara, 2022-10-18:
WiiPointTracker does not appear in Opentrack-2022.3.1 (but WiiPointTracker does appear in Opentrack-2022.3.0)
Windows 7.

```
E:\Tools\Opentrack\OT-2022.3.1-portable>opentrack

E:\Tools\Opentrack\OT-2022.3.1-portable>
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" faile
d: "Cannot load library E:\\Tools\\Opentrack\\OT-2022.3.1-portable\\modules\\\\o
pentrack-tracker-kinect-face.dll: DEBUG [F:\dev\opentrack\api/plugin-support.hpp
:106]: library "wii" failed: "Cannot load library E:\\Tools\\Opentrack\\OT-2022.
3.1-portable\\modules\\\\opentrack-tracker-wii.dll: DEBUG [F:\dev\opentrack\open
track\main-window.cpp:1030]: no preset dir

```

oliverCJ, 2022-10-19:
same error in Opentrack-2022.3.1
system: win10
`
DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: DEBUG [F:\dev\opentrack\api/plugin-support.hpp:106]: library "wii" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-wii.dll: DEBUG [F:\dev\opentrack\opentrack\main-window.cpp:1030]: no preset dir
libpng warning: iCCP: cHRM chunk does not match sRGB
`

sthalik, 2022-10-19:
See if a clean recompile helps https://www.dropbox.com/s/kmq1sz04bpyn1au/opentrack-test-20221019_01.7z

GO63-samara, 2022-10-19:
This is good!
Wii appeared and works in opentrack-test-20221019_01
Thanks, @sthalik

sthalik, 2022-10-19:
Thanks! I'm releasing a new build immediately.

sthalik, 2022-10-19:
@GO63-samara please test the 2022.3.2 release. I've had to recompile it to apply the version name for the titlebar.

GO63-samara, 2022-10-19:
Checked OT-2022.3.2. The Wii is on the list. Thank you!

closed 2022-10-23 by GO63-samara