opentrack/opentrack #1558: Updated Fedora build guide as current one is out of date.

tsoulard, 2022-10-22:
While trying to build opentrack on fedora i noticed the build guide was out of date. I'd like to update myself but obviously don't have permissions so I've induced the guide below. 

# For Wiki
To start we need to add some dependencies. Please run the following command.

```bash
sudo dnf install cmake git qt5-qttools-devel qt5-qtbase-private-devel procps-ng-devel opencv-devel wine-devel wine-devel.i686 glibc-devel.i686
```

Now clone the `opentrack` repo 
```bash
git clone https://github.com/opentrack/opentrack
cd opentrack/
```

To build run the below
```bash
mkdir build
cd build
cmake ..
```
To enable proton run `ccmake .` and select `SDK_WINE` using the arrows keys. Then press enter to change the option to on and press `c` to reconfigure and then `g` to generate and exit.

Continuing
```bash
make 
make install
cd install/bin
# to run opentrack run the below command
./opentrack
```

Now you can remove some of the dependencies if you don't need them. 
```bash
sudo dnf remove cmake git qt5-qttools-devel qt5-qtbase-private-devel wine-devel wine-devel.i686 glibc-devel.i686
```
> Double check you're not removing any software here that you rely on for work or hobby projects such as `git` or `cmake`

erlking, 2023-02-22:
thanks, fixed some of my missing dep issues!

EnfrigFalco, 2025-01-14:
You still need these dependencies:

`sudo dnf install qt5-qtserialport-devel onnxruntime-devel`

I am pretty sure onnxruntime-devel is needed for the new neuraltracker