opentrack/opentrack #1559: Opentrack 2022.3.1 and .2 fails to build on Linux (Fedora 36)  with neuralnet tracker enabled.

XenoPL, 2022-10-25:
As per title attempt to build when path to onnxruntime is provided with error:

```
[ 84%] Building CXX object tracker-neuralnet/CMakeFiles/opentrack-tracker-neuralnet.dir/ftnoir_tracker_neuralnet.cpp.o
In file included from ~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.cpp:8:
~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.h:30:10: fatal error: onnxruntime/core/session/onnxruntime_cxx_api.h: Nie ma takiego pliku ani katalogu
   30 | #include <onnxruntime/core/session/onnxruntime_cxx_api.h>
```

Opentrack 2022.3.0 compiles just fine. Diff on tracker-neuralnet/ftnoir_tracker_neuralnet.h file between 2022.3.0 and 2022.3.2:

```
diff ~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.h ~/tmp/opentrack/opentrack-opentrack-2022.3.0/tracker-neuralnet/ftnoir_tracker_neuralnet.h 

30c30
< #include <onnxruntime/core/session/onnxruntime_cxx_api.h>
---
> #include <onnxruntime_cxx_api.h>

```
If I change the include back to the way it is in 2022.3.0 it goes a bit further but stops with another error:

```
~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.cpp: In constructor ‘neuralnet_tracker_ns::PoseEstimator::PoseEstimator(Ort::MemoryInfo&, Ort::Session&&)’:
~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.cpp:420:62: error: ‘struct Ort::Session’ has no member named ‘GetOutputName’
  420 |     if (session_.GetOutputCount() >= 3 && "box"s == session_.GetOutputName(2, allocator_))
      |                                                              ^~~~~~~~~~~~~
~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.cpp:447:41: error: ‘struct Ort::Session’ has no member named ‘GetInputName’
  447 |         input_names_.push_back(session_.GetInputName(i, allocator_));
      |                                         ^~~~~~~~~~~~
~/tmp/opentrack/opentrack-opentrack-2022.3.2/tracker-neuralnet/ftnoir_tracker_neuralnet.cpp:451:42: error: ‘struct Ort::Session’ has no member named ‘GetOutputName’
  451 |         output_names_.push_back(session_.GetOutputName(i, allocator_));
      |                                          ^~~~~~~~~~~~~
make[2]: *** [tracker-neuralnet/CMakeFiles/opentrack-tracker-neuralnet.dir/build.make:90: tracker-neuralnet/CMakeFiles/opentrack-tracker-neuralnet.dir/ftnoir_tracker_neuralnet.cpp.o] Błąd 1
make[1]: *** [CMakeFiles/Makefile2:2801: tracker-neuralnet/CMakeFiles/opentrack-tracker-neuralnet.dir/all] Błąd 2
make: *** [Makefile:136: all] Błąd 2

```

DaWelter, 2022-10-25:
`#include <onnxruntime_cxx_api.h>` is what I would expect. The other path was perhaps automatically added. VSCode does that sometimes uncalled for .. @sthalik ?

It should work with your change. Aren't there other hints?

XenoPL, 2022-10-25:
Ok, update, I can succesfully build 2022.3.2 when said include is reverted back to 2022.3.0 state ie:
`#include <onnxruntime_cxx_api.h>`
and using onnxruntime 1.12.1

However there'll be an issue with 1.13.1 onnxruntime where  compilation fails with 
struct Ort::Session’ has no member named ‘GetOutputName’

sthalik, 2022-10-26:
@DaWelter, the build system is passing `-I F:\dev\opentrack\..\opentrack-depends\onnxruntime-1.10.0\include` and no other paths. The settings look normal:

```console
dev/opentrack/build % grep -i ^onnx CMakeCache.txt
ONNXRuntime_DIR:INTERNAL=F:/dev/opentrack/../opentrack-depends/onnxruntime-1.10.0
ONNXRuntime_INCLUDE_DIR:INTERNAL=F:/dev/opentrack/../opentrack-depends/onnxruntime-1.10.0/include
ONNXRuntime_LIBRARY:INTERNAL=F:/dev/opentrack/../opentrack-depends/onnxruntime-1.10.0/lib/onnxruntime.lib
ONNXRuntime_RUNTIME:INTERNAL=F:/dev/opentrack/../opentrack-depends/onnxruntime-1.10.0/bin/onnxruntime.dll
```

What are yours?

sthalik, 2022-10-26:
It should work for you now.

@XenoPL, you need to replace

```c++
if (session_.GetOutputCount() >= 3 && "box"s == session_.GetOutputName(2, allocator_))
```

with:

```
const auto ptr = session_.GetOutputNameAllocated(2, allocator_);
if (session_.GetOutputCount() >= 3 && "box"s == &*ptr)
```

similarly:

```c++
input_names_.push_back(session_.GetInputName(i, allocator_));
```

with

```c++
const auto ptr = session_.GetInputNameAllocated(i, allocator_);
input_names_.push_back(&*ptr);
```

sthalik, 2022-10-26:
Maybe you can even replace:

```c++
session_.GetOutputName(2, allocator_)
```

with:

```c++
&*session_.GetOutputNameAllocated(2, allocator_)
```

XenoPL, 2022-10-26:
Thanks, I'll try to build OT with onnxruntime 1.13 in the evening.

Sp far OT + onnx 1.12 works lovely. I think neuralnet tracker is on par (if not better) than PointTracker. It seems I can put freetrack kit to the closet. Great job guys.

DaWelter, 2022-11-01:
@sthalik In the linux distribution the headers are directly in the include dir like so
[image]
On windows I just copy pasted the files so the directory structure is the same ...

DaWelter, 2022-11-01:
I guess we need a better find script for cmake. I'll try to "borrow" one from the internet ;-)

sthalik, 2022-11-01:
But this flaky script is the devil we know.

```cmake
    if(EXISTS "${ONNXRuntime_INCLUDE_DIR}/onnxruntime/core/session/.")
        include_directories("${ONNXRuntime_INCLUDE_DIR}/onnxruntime/core/session")
    endif()
```

DaWelter, 2022-11-01:
Let's discuss it there https://github.com/opentrack/opentrack/pull/1562

turmoni, 2023-01-06:
I've also come across this build issue, albeit now in `tracker-neuralnet/model_adapters.cpp` (I assume there's been some refactoring, but this assumption may be wrong).

I've tried using the approach suggested above (re: changing GetOutputName to GetOutputNameAllocator, etc.) and this does build, but tracking doesn't work - I only get a blue rectangle around my face, and my output is spammed with:

```
DEBUG [/[...]/opentrack/tracker-neuralnet/model_adapters.cpp:325]: Failed to run the model:  Invalid Output Name:x
```

I did a tiny bit of looking to see if I could possibly figure out what this means and how to perhaps fix it, but I've realised this is going to require me actually understanding what onnxruntime is and the high level of how it works to have an idea as to what the problem is, so it might take a while.

turmoni, 2023-01-06:
Well I seem to have made it work well enough _for me_ (caveat: haven't actually tried it in-game yet, but the octopus moves appropriately) by hacking in the output names, so my relevant complete changes against 8369f50018f796bb1da957b4404c01580102f2fb are:
```
diff --git a/tracker-neuralnet/model_adapters.cpp b/tracker-neuralnet/model_adapters.cpp
index af599321..96dad832 100644
--- a/tracker-neuralnet/model_adapters.cpp
+++ b/tracker-neuralnet/model_adapters.cpp
@@ -217,12 +217,12 @@ PoseEstimator::PoseEstimator(Ort::MemoryInfo &allocator_info, Ort::Session &&ses
     qDebug() << "Pose model outputs (" << session_.GetOutputCount() << "):";
     for (size_t i=0; i<session_.GetOutputCount(); ++i)
     {
-        const char* name = session_.GetOutputName(i, allocator_);
+        const auto name_ptr = session_.GetOutputNameAllocated(i, allocator_);
         const auto& output_info = session_.GetOutputTypeInfo(i);
         const auto& onnx_tensor_spec = output_info.GetTensorTypeAndShapeInfo();
-        auto my_tensor_spec = understood_outputs.find(name);
+        auto my_tensor_spec = understood_outputs.find(&*name_ptr);
 
-        qDebug() << "\t" << name << " (" << onnx_tensor_spec.GetShape() << ") dtype: " <<  onnx_tensor_spec.GetElementType() << " " <<
+        qDebug() << "\t" << &*name_ptr << " (" << onnx_tensor_spec.GetShape() << ") dtype: " <<  onnx_tensor_spec.GetElementType() << " " <<
             (my_tensor_spec != understood_outputs.end() ? "ok" : "unknown");
 
         if (my_tensor_spec != understood_outputs.end())
@@ -230,7 +230,7 @@ PoseEstimator::PoseEstimator(Ort::MemoryInfo &allocator_info, Ort::Session &&ses
             TensorSpec& t = my_tensor_spec->second;
             if (onnx_tensor_spec.GetShape() != t.shape || 
                 onnx_tensor_spec.GetElementType() != Ort::TypeToTensorType<float>::type)
-                throw std::runtime_error("Invalid output tensor spec for "s + name);
+                throw std::runtime_error("Invalid output tensor spec for "s + &*name_ptr);
             output_val_.push_back(Ort::Value::CreateTensor<float>(
                 allocator_info, t.buffer, t.element_count, t.shape.data(), t.shape.size()));
             t.available = true;
@@ -240,7 +240,7 @@ PoseEstimator::PoseEstimator(Ort::MemoryInfo &allocator_info, Ort::Session &&ses
             // Create tensor regardless and ignore output
             output_val_.push_back(create_tensor(output_info, allocator_));
         }
-        output_names_.push_back(name);
+        output_names_.push_back(&*name_ptr);
     }
 
     has_uncertainty_ = understood_outputs.at("rotaxis_scales_tril").available ||
@@ -272,7 +272,8 @@ PoseEstimator::PoseEstimator(Ort::MemoryInfo &allocator_info, Ort::Session &&ses
 
     for (size_t i = 0; i < session_.GetInputCount(); ++i)
     {
-        input_names_.push_back(session_.GetInputName(i, allocator_));
+        const auto name_ptr = session_.GetInputNameAllocated(i, allocator_);
+        input_names_.push_back(&*name_ptr);
     }
 
     assert (input_names_.size() == input_val_.size());
@@ -308,6 +309,8 @@ std::optional<PoseEstimator::Face> PoseEstimator::run(
 
     Timer t; t.start();
 
+    const char* fixoutput[] = {"pos_size", "quat", "box"};
+
     try
     {
         session_.Run(
@@ -315,7 +318,8 @@ std::optional<PoseEstimator::Face> PoseEstimator::run(
             input_names_.data(), 
             input_val_.data(), 
             input_val_.size(), 
-            output_names_.data(), 
+            fixoutput,
+//            output_names_.data(), 
             output_val_.data(), 
             output_val_.size());
     }
```

This is obviously not a _real_ fix

l3ackEyeD, 2023-03-12:
Well I managed to build it with according to instructions above. 
But when I load OT it fails to load Neuralnet
(/home/phil/opentrack/build/install/bin/../libexec/opentrack//opentrack-tracker-neuralnet.so: undefined symbol: omp_set_num_threads)"

XenoPL, 2023-03-25:
+ 1 OpenTrack 2023.1.0 won't build with onnx 1.13 and later.
Error messages:
`/home/Xeno/tmp/opentrack/opentrack-opentrack-2023.1.0/tracker-neuralnet/model_adapters.cpp: In constructor ‘neuralnet_tracker_ns::PoseEstimator::PoseEstimator(Ort::MemoryInfo&, Ort::Session&&)’:
/home/Xeno/tmp/opentrack/opentrack-opentrack-2023.1.0/tracker-neuralnet/model_adapters.cpp:220:37: error: ‘struct Ort::Session’ has no member named ‘GetOutputName’
  220 |         const char* name = session_.GetOutputName(i, allocator_);
      |                                     ^~~~~~~~~~~~~
/home/Xeno/tmp/opentrack/opentrack-opentrack-2023.1.0/tracker-neuralnet/model_adapters.cpp:275:41: error: ‘struct Ort::Session’ has no member named ‘GetInputName’
  275 |         input_names_.push_back(session_.GetInputName(i, allocator_));
`

XenoPL, 2023-03-25:
Ok undefined symbol: omp_set_num_threads issue solved by extra cmake flag:  -DCMAKE_CXX_FLAGS="-fopenmp".

l3ackEyeD, 2023-03-27:
Can confirm that it compiles and starts on onnx 1.13, with the changes to CMakeList.txt above.
Sadly I didnt encouter a new error while trying to run Neuralnet. 
```
DEBUG [/home/phil/opentrack/tracker-neuralnet/model_adapters.cpp:328]: Failed to run the model:  Invalid Feed Input Name:@�=�����v?�na<���`�c���@@�=���^K��%�нk9=ZB�==��=$#=Tx�=�g�<;�S>��Q��]
```

l3ackEyeD, 2023-03-27:
Can confirm that opentrack builds with onnx >=1.13 with the changes from [XenoPL](https://github.com/XenoPL).
It starts and Neuralnet is beeing recognized. Still needs the fixes mentioned by [turmoni](https://github.com/turmoni) above. Otherwise output is beeing spammed with invalid Feed input.

XenoPL, 2023-03-28:
+1 OT with extra -fomp flag and turmoni changes builds ok, but neuralnet tracker fails to work with message above.

zhaozhixu, 2023-04-06:
@XenoPL @sthalik @l3ackEyeD 
`Failed` to run the model:  Invalid Feed Input Name:` is because that the string returned by `GetInputNameAllocated` is destroyed beyond its scope

 I have created a patch #1638 that saves the names in `std::string`s, which resolves that problem. Also I wraps the onnxruntime calling code in functions with version macro tests.

I have tested it on my Fedora 36. Please test on your platforms and I'm glad to have some reviews.

XenoPL, 2023-04-07:
With latest changes merged OT 2023.1.0.6 has been successfully build with onnx runtime 1.14.1.

l3ackEyeD, 2023-04-11:
Can confirm the report from @XenoPL above. Builds fine for me as well OT 2023.1.0.6 + onnx runtime 1.14.1. Issue seems fixed

closed 2023-04-26 by XenoPL