opentrack/opentrack #1560: Mouse emulation (OT in a Linux environment)

gerrygk, 2022-10-26:
Hi all,

in issue #761 it is mentioned that mouse emulation is possible with OT. Unfortunately, I didn't find such an output mode. (The ones I find are FlightGear, libevdev joystick receiver and UDP over network.) I'm wondering how to apply that.

Any help available?

Thx in advance

forbesmyester, 2022-10-27:
I'm looking for this too. Think it's not (currently) possible but I'm looking for something that makes mouse look in American Truck Simulator possible. Joystick support seems to be only relative, not absolute, so it's not working...

I thought Options -> Relative Translation would fix it, but it doesn't seem to do anything.

sthalik, 2022-10-27:
You can already use virtual joystick output in ETS and its sequels.

forbesmyester, 2022-10-27:
However centering the joystick does not center the view, and holding the joystick left (stationary) continues to look more and more left.

If that was not clear, another way of explaining, left on the joystick causes the head to continue to rotate left at, say 10 degrees a second. not to move to say 30 degrees and stay there.

sthalik, 2022-10-27:
Can't you add the evdev virtual joystick as a second joystick whose only purpose is to provide head pose data?

gerrygk, 2022-10-27:
I already tried OT with virt joy output mode together with IL-2 1946, but it didn't work fine. (I got the same results as forbesmyester.) In this game, one can use the mouse to take a look around in the cockpit. So I think it would be valuable if a virtual mouse output would be available.

sthalik, 2022-10-27:
IL-2 1946 runs through Proton so you can use Wine output.

Mouse output is always an inferior solution (drift and range issues) so it's not a priority.

forbesmyester, 2022-10-27:
I'm evaluating buying some tracking hardware.

So, I'm using a wheel (Logi G25) and pedals (Fanatec).

Binding the wheel access to linux joypad input for yaw causes the octopus to turn his head.

I'd then pick libevdev output and I can see in `jstest-gtk` that the output of that joystick mirrors what I do with the joystick.

Picking either the physical or virtual (opentrack) joystick casues the view in American Truck Simulator to turn at a speed defined by the position of a wheel (10 degrees a second), rather than turn to an angle defined by that position of the wheel (30 degrees).

It does seem that  American Truck Simulator should take the axis as an angle to gaze at (or have that option).

gerrygk, 2022-10-27:
@sthalik: I always try to avoid using wine as it seems to have some security issues.

gerrygk, 2022-10-27:
> Can't you add the evdev virtual joystick as a second joystick whose only purpose is to provide head pose data?

I'm gonna give it a try, but I doubt that it's being recognised by IL-2.

gerrygk, 2022-10-27:
As I thought, unfortunately it doesn't work.
Maybe there's a problem with the libevdev package. The /dev/input/uinput file (mentioned in the libevdev options of OT) is there.

sthalik, 2022-10-27:
Did you install libevdev?

gerrygk, 2022-10-27:
Definitely, an alternative for Freetrack 2 output is missing in Linux...

sthalik, 2022-10-27:
You should get `proto/libevdev` to output on stderr when it fails.

gerrygk, 2022-10-27:
> Did you install libevdev-dev?

Yes, both libevdev2 and libevdev-dev.

gerrygk, 2022-10-27:
> You should get `proto/libevdev` to output on stderr when it fails.

Aha, what to do now?

sthalik, 2022-11-13:
> Aha, what to do now? Unfortunately, I'm not really familiar with this...

Open a terminal, such as Konsole, gnome-terminal or xterm. Execute `opentrack` from where you installed it and look at what it spits out.

gerrygk, 2022-11-14:
When doing so, I only get the message
"DEBUG [/home/gerry/Programme/opentrack-opentrack-2022.3.2/opentrack/main-window.cpp:1030]: no preset dir

gerrygk, 2022-11-14:
OT is running well btw

sthalik, 2022-11-14:
Which IL-2 game are you talking about? Also, is it a native version or one running under Wine, or under Proton?

gerrygk, 2022-11-14:
I've installed IL-2 1946 in Steam, and I'm using Proton 7 to run it

gerrygk, 2022-11-14:
My OS is Mint 21 btw

gerrygk, 2022-11-22:
> As I thought, unfortunately it doesn't work. Maybe there's a problem with the libevdev package. The /dev/input/uinput file (mentioned in the libevdev options of OT) is there.

I was wrong there... There's only a /dev/uinput device, no /dev/input/uinput.

sthalik, 2022-11-22:
Why not use the Wine output method though?

gerrygk, 2022-11-23:
Recently I brought myself to try the wine output, and did this by following the installation instruction posted in 

https://volatilemark.github.io/opentrack-wine-guide/

but ran into problems because obsolete wine versions are needed for this purpose (5.6 or older). These don't seem to go with my current OS (Mint 21).

gerrygk, 2022-11-23:
I now installed Wine (v7.0.1) now from WineHQ as it is described there. Trying to compile OT with SDK_WINE activated and SDK_WINE_PREFIX (= /bin) set in CMakeCache.txt, I get the error message

>>
[ 66%] Generating opentrack-wrapper-wine.exe.so
In file included from /home/gerry/Programme/opentrack-opentrack-2022.3.2/proto-wine/opentrack-wrapper-wine-main.cxx:15:
/home/gerry/Programme/opentrack-opentrack-2022.3.2/compat/shm.h:10:10: fatal error: windows.h: Datei oder Verzeichnis nicht gefunden
   10 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
winegcc: /usr/bin/g++ failed
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:75: proto-wine/opentrack-wrapper-wine.exe.so] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:2279: proto-wine/CMakeFiles/wine-wrapper.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
<<

so the windows.h file could not be found.

gerrygk, 2022-11-23:
Solved this by additional installation of the wine-stable-dev (v7.0.1) package, now "make" succeeded...

gerrygk, 2022-11-23:
The "make install" step also worked now.

gerrygk, 2022-11-23:
Now I configured the wine output in OT (wine settings):
wine variant = Proton 7.0, and set the Steam app ID.

While starting OT, I get the error msg:
***
Error occurred while loading protocol Wine -- Windows layer for Unix
Couldn't find a Steam runtime.
***

gerrygk, 2022-11-25:
I think this OT/wine case is off topic here. Should open a new issue...

gerrygk, 2022-12-01:
Now continuing with issue #1580...

closed 2022-12-01 by gerrygk

digitaltrails, 2022-12-03:
> Hi all,
> 
> in issue #761 it is mentioned that mouse emulation is possible with OT. Unfortunately, I didn't find such an output mode. (The ones I find are FlightGear, libevdev joystick receiver and UDP over network.) I'm wondering how to apply that.
> 
> Any help available?
> 
> Thx in advance

I had recently been looking for a way to get some head tracking going with IL-2 BOS in Linux Steam Proton.  As part of that research I've written a [gist that converts opentrack UDP-output to Linux-HID mouse events](https://gist.github.com/digitaltrails/299ce6c4c970eef1f46a5bcbbe1c69f2) using evdev/libevdev.   It's written in python, there's not much code to it, feel free to tinker with it.

Given it's injecting mouse events into the kernel hid subsystem layer, it can't fail to work with Steam Proton IL-2 BOS.  

It's a bit jerky, but that may, or may not, be resolvable with some tuning, either in opentrack, upstream in the tracking device, or in my code. 

Outside of IL-2 BOS, it seems OK, I imagine it could at least be useful for non-gaming purposes.

Inside IL-2 BOS, I wonder if the x, y, z coupled with the yaw, pitch and roll, might be used to figure out the head is pointing straight ahead and automatically issue a recenter viewpoint.  I find drifting off centre and needing to do a manual reset is a bit annoying.  (Originally I coded it to work off of x,y,z, but then I found it worked better to use yaw, pitch and roll.)

I also haven't had much success in converting the z-axis into a useful mouse-wheel zoom, it could be a tuning issue, or maybe just using z-values isn't sufficient.  

I just wrote it yesterday, so it's not like I've really studied the problem.  At this point  I don't find it superior to strapping a cheap Aliexpress gyro mouse pointer to my headphones.

gerrygk, 2022-12-05:
Hi @digitaltrails,

thanks for your efforts, I tried the script with Aruco input in OT and it works quite decently. Unfortunately I have similar problems (centering issues, movement is not really smooth). For me, the EWMA filter doesn't work as good as Accela.

The scale factor of 35 seems to be a bit too much, though.

But anyway, that's a good first try in my opinion.

digitaltrails, 2022-12-05:
> Hi @digitaltrails,
>
> thanks for your efforts, I tried the script with Aruco input in OT and it works quite decently. Unfortunately I have similar problems (centering issues, movement is not really [...]

Hi @gerrygk,

Thanks for the feedback, much appreciated.

The default scale factor is OK for use in a windowing environment.  I had hoped the mouse-emulation might be useful for Linux folk that need a head mouse outside of gaming.  At -f 35, it seems quite usable for that.   But, yes,  I find I have to lower it to 10 when in IL-2 BOS.   At 10 I found EWMA seemed smoother, but maybe I was imagining it, because it's still not smooth.

I tried adding moving-average smoothing to the python, but it had no substantial effect.  Then I measured the time between outputs and found it to be 4 milliseconds.  Four milliseconds is a worry because 1000 MHz mice are said to be about right for gaming.   

My suspicion is that either the UDP feed from opentrack, or the python reading it, is too slow.   I need to measure the two separately.  If the python implementation isn't the bottleneck, it's possible it could wake up between real-events and interpolate more events between each of the 4 ms events (basically keep up the momentum of the mouse moving in a particular direction).

If the python is the bottleneck, it's not a very large program, it might be converted to C/C++, or recoded as an actual opentrack output.

The centering issue is a difficult one to solve, because a mouse is a relative-position event device.  Rather than injecting mouse events, it might be better to create a HID layer virtual joystick (joysticks are absolute-position event driven).  I don't think the evdev package can create virtual devices.  

I've discovered there is alternative [libevdev](https://python-libevdev.readthedocs.io/en/latest/index.html) python package that can create virtual HID devices.   Perhaps it could be used to create one or more virtual joysticks and bind various joystick axes to the IL-2 BOS head axes (I believe that IL-2 allows multiple controller devices).  If that panned out, the opentrack x, y, z, yaw, pitch, roll could all be mapped, thus allowing for proper head tracking.  

The first thing to find out is whether IL2-BOS can see a virtual joystick.  Given the virtual joystick would be a kernel HID subsystem device, I don't see why not.  Of course, like the mouse, this may still be too slow/jerky, but if I can knock up a proof of concept in python,  the python effort could be recoded in C or C++ (it's more than twenty years since I last did any substantial C/C++ coding - so possibly not by me, but maybe).  

So that's a few things to look at on future rainy days.  I'm not claiming an exclusive on these ideas, if anyone else feels inspired, please feel free to jump in.

digitaltrails, 2022-12-07:
> > Hi @digitaltrails,
> > thanks for your efforts, I tried the script with Aruco input in OT and it works quite decently. Unfortunately I have similar problems (centering issues, movement is not really smooth). For me, the EWMA filter doesn't work as good as Accela.
> > The scale factor of 35 seems to be a bit too much, though.
> > But anyway, that's a good first try in my opinion.
> 
> Hi @gerrygk,
> 
> Thanks for the feedback, much appreciated.
> 
> ...
> 
> My suspicion is that either the UDP feed from opentrack, or the python reading it, is too slow. I need to measure the two separately. If the python implementation isn't the bottleneck, it's possible it could wake up between real-events and interpolate more events between each of the 4 ms events (basically keep up the momentum of the mouse moving in a particular direction).
> 

I updated the gist.  It turns out it was waiting for input from opentrack (for 4 ms).  So presuming 1000 MHz is the normal speed for a mouse, I'm now using a non-blocking read that times out in 1 ms, at which point I use the last mouse move to fill, and keep doing that until new data eventually arrives.   I think it looks smoother (with smoottrack on Android, EWMA in opentrack, scale factor 10-15 in opentrack-mouse).

> I've discovered there is alternative [libevdev](https://python-libevdev.readthedocs.io/en/latest/index.html) python package that can create virtual HID devices. Perhaps it could be used to create [...]

Turns out it was a permission issue that prevented evdev from registering new virtual devices, so I don't need to switch python libraries.  This evening I managed to create a new virtual-joystick.  Steam appears to see it.  IL-2 seems to also to see it - under key mappings.   But for some reason I can't map it.  IL-2 seems to know about different brand/models of joysticks, maybe I need to define it in some config file somewhere.   Early days yet.

Another approach is I could just convert the opentrack feed into keystrokes which IL2 would not be able to distinguish from the actual keyboard. Surely that should work (but centering would again be an issue).

digitaltrails, 2022-12-11:
> 

> I updated the gist. It turns out it was waiting for input from opentrack (for 4 ms). So presuming 1000 MHz is the normal speed for a mouse, I'm now using a non-blocking read that times out in 1 ms, [...]

I gone about as far as I can without investing in some better hardware.  

- I already added the above fill/interpolation during the 4 ms interval,...
- I've also added a low pass filter.  
- I've added an attempt at auto-centering. 
- I've gotten the z-axis head-zoom roughly usable.

It's still not that smooth when using the `smoothtrack` Android app.  It could be the combo of hardware, phone, PC, Wifi router, or it could be my code.     When I test using a locally connected joystick as the opentrack-input, the resulting opentrack-mouse movement seems quite smooth.   

I can see that the joystick moves often come down to 0 or 1 unit of movement.   Perhaps I should simulate acceleration/deceleration with sequences of zeros and ones.  But the joystick-test input resulted in quite smooth output. Puzzled.

I did write a joystick emulator: `opentrack-stick`.  But I found my knowledge of IL-2 BOS isn't sufficient to get that working. I managed to toast the input configs and had to go to backup to recover them.  I think the game has some ideas of models of sticks - maybe I need to emulate a particular make and model - or maybe I just need to know the right IL-2 BOS magic.

In case someone might find them useful, I've pushed both of these efforts into a github repository called [opentrack-opal  - _Opentrack Python And Linux_](https://github.com/digitaltrails/opentrack-opal).   I'll likely set the  code aside until I obtain an alternative to `smoothtrack` and/or some feedback from more users.  Smoothtrack is draining my Android phone's battery way too fast.  I'm going to stop using that.  Maybe I will look to building my own LED based hat.  My use of opentrack is parked for now.

In the meantime I wear a `G11 Air Mouse RGB Backlit Gyro 2.4G Universal Remote Control`  under by headphone headband ($8 from aliexpress).  I jam down RGB key to stop it from falling asleep.  It's silky smooth.  But I definitely need to allocate a joystick buttons to centering and zoom.

digitaltrails, 2022-12-13:
> I did write a joystick emulator: `opentrack-stick`. But I found my knowledge of IL-2 BOS isn't sufficient to get that working. I managed to toast the input configs and had to go to backup to recover [...]

I go the joystick working in IL-2 BoX (see github for details).  With smoothing implemented, it's not too bad.  There
's absolutely no center-drifting (with smoothtrack).  I suspect the smoothing algorithm could be further improved, but it seems relatively usable (pitch and yaw only for now).

gerrygk, 2022-12-14:
It would be useful to know how the freetrack protocol in detail works as output in OT, as it seems to be compatible with all IL-2 game types and versions.

> I go the joystick working in IL-2 BoX (see github for details).
Where exactly? I couldn't find it with a general search in github.

digitaltrails, 2022-12-14:
> It would be useful to know how the freetrack protocol works as output in detail, as it seems to be compatible with all IL-2 game types and versions.
> 
> > I go the joystick working in IL-2 BoX (see github for details).
> 
> Where exactly? I couldn't find it with a general search in github.

I ment, I got **my own python joystick emulator** to work:  [opentrack-stick](https://github.com/digitaltrails/opentrack-opal/blob/ef56cc658295527fa61fb0270fda30dfdf2e22cf/opentrack-stick.md).  It emulates a Microsoft X-Box 360 controller.

I need to improve it.  Add the ability to direct x and y to the hat (because IL-2 only seems to uses keys/hat for head movement). I would like to make setting it up easier - some how.

digitaltrails, 2022-12-14:
> It would be useful to know how the freetrack protocol works as output in detail, as it seems to be compatible with all IL-2 game types and versions.
> ...

The problem I have with using the existing IL-2 BoX supported protocols, is that it seems incredibly hard, or hit and miss,  to get them working in Linux+Steam/Proton.  I have not come across any description of how to get them working in Linux+Steam+Proton.  Plus people seem to be taking all kinds of weird approaches, using flatpack, lutris, different distro's, and magic-chicken-feet, ...

I figured the easiest way forward is just to emulate a second stick controller.  That side steps any special IL-2 BoX, Steam+Proton, Wine, issues.  It narrows things down to just getting more than one stick controller working.  

More than one controller is not supposed to be an issue.  My guess is that past trouble I had with trying to use a stick approach, is that I need to emulate a well known controller,  so I dumped the capabilities of my Nintendo Pro Controller, which Linux evdev dumped out as a Microsoft X-Box 360 pad, I guess that is a common as it gets.

Now my IL-2 Box (OpenSUSE, official OpenSUSE Steam), sees my Logitech stick, and this virtual Microsoft-controller.  Having implemented something that looks like a known controller, IL-2 BoX seemed to recognise it with no issues, and it was happy to let me bind it's axes.   

The next problem was how use wobbly head movements to set IL-2 BoX "Key Mappings".  The the trick was to work on one axis at a time, turning off the other head-axes by setting them  to a flat mapping graph in opentrack.   To map an axis, IL-2 needs to see it move across its entire range.  The opentrack curve needs to be adjusted so that the output reaches it's maximum.  By default, opentrack's curves output perhaps half the range, this is not enough for IL-2, it's "Key" mapper just ignores the input.

After setting up head-pitch and head-yaw, tracking worked, but it was jerky, I had to code up some smoothing, a figure out some smoothing parameters.   Now it actually appears usable,  preferable to any mouse based approach I tired, worth investing some more time it at any rate.

This works well for head pitch and yaw.  My guess is that the head-tracking specific protocols would be better at the head/body moving left/right/up/down in the cockpit.  For IL-2  I'll have implement those as hat moves, which only have limited values -1, 0, 1.