opentrack/opentrack #1564: No tracking data from SteamVR with null driver

nexrobo, 2022-11-01:
Same behavior as in #1486

Opentrack version 2022.3.2
SteamVR version 1.24.6
Using vive tracker 3.0, no headset connected (null driver)

when starting tracking while steamvr is running and the tracker is on i get:
DEBUG [F:\dev\opentrack\tracker-steamvr\steamvr.cpp:149]: steamvr: no valid pose from device 1 valid false connected true
otherwise a whole lot of:
DEBUG [F:\dev\opentrack\tracker-steamvr\steamvr.cpp:60]: steamvr: no device with index
Inside steamvr (specifically the "headset window") the vive tracker gets tracked as expected.

As in #1486 i do remember it working while the headset was connected (a while ago on an older version of opentrack and probably steamvr), sadly i hasn't started working magically :) the link in #1483 isn't working anymore.

sthalik, 2022-11-02:
How is it meant to work when no device is connected?

nexrobo, 2022-11-02:
Sorry for the confusion, i should have been more precise:

When i start tracking _the first time_ after opening Opentrack there are how ever many lines of:
DEBUG [F:\dev\opentrack\tracker-steamvr\steamvr.cpp:60]: steamvr: no device with index
ending with:
DEBUG [F:\dev\opentrack\tracker-steamvr\steamvr.cpp:149]: steamvr: no valid pose from device 1 valid false connected true

When i subsequently stop and start the tracking (without restarting Opentrack) i only get the first error (probably the same number of times like before) without the second.

All of course while steamvr is running, the tracker is on and being tracked. Also the tracker is selected in Opentracks input setting.
Hope that clears things up, sorry again

Frosty704, 2022-12-29:
if you havent figured it out yet try right clicking steamvr -> developer -> developer settings -> room and tracking -> quick calibrate