opentrack/opentrack #1565: Trojan threat detected

Sheridan-Tech, 2022-11-01:
BitDefender flagged opentrack-2022.3.2-win32-portable\install\modulesps3eye-subprocess.exe as a generic Trojan.

sthalik, 2022-11-02:
It's not a trojan, as I built it along with all dependencies. You can keep the file deleted as long as you don't need the PS3 Eye open driver.

Sheridan-Tech, 2022-11-03:
> It's not a trojan, as I built it along with all dependencies. You can keep the file deleted as long as you don't need the PS3 Eye open driver.

I think it's risky to assume its a false positive simply because it was built with all the dependencies. Especially because I downloaded previous versions of the release just after that and that file was not flagged in those. So there is some difference in that executable between releases that matches the trojan signature.

sthalik, 2022-11-03:
There is a difference. I updated libusb.

Maskuh, 2023-04-19:
> BitDefender flags \opentrack-2022.3.2-win32-portable\install\modules\ps3eye-subprocess.exe as a generic Trojan.

Mcafee Also Flags this as a Trojan.