opentrack/opentrack #1570: neuralnet tracker - Z  does not change

jippy13011, 2022-11-09:
Hi,

I tried the last version of opentrack 2022.3.2 with neuralnet /feedtrack/ Accela and noticed that z axis is fixed. The zoom does not work.

I noticed too that, activating the csv data file loging at the start solves this problem...

Does anyone has a solution without activating the log file?

sthalik, 2022-11-09:
Pinging @DaWelter.

sthalik, 2022-11-09:
> Beware that the Z scaling is a bit too large by a x2 to x3 factor in neuralnet, so when I stand around 70cm from my Webcam, I get Z values around 150 to 180

Use the FOV setting. It doesn't have to be the real value.

jippy13011, 2022-11-10:
Thank you for answering.
@Pieloth 
francais ? I remarked that the Z value of the output is stuck at the max value, although  the z value in the raw tracker data is changing correctly ie the soft works to detect my z motion.
few minutes later
I just made another try and finally it works by setting z max input 150, z max output 300cm (in mapping).
Thanks !!

DaWelter, 2022-11-11:
> Beware that the Z scaling is a bit too large by a x2 to x3 factor in neuralnet,
That should be fixed in the next release.
I noticed the stuck pose as well. But it's only on startup. After centering it works normally.

sthalik, 2022-11-11:
@DaWelter, stuck pose is likely related to the camera's exposure settings on the few initial frames. You could try ignoring some 30 frames at startup.

Pieloth, 2022-11-12:
> @Pieloth francais ? 

Oui !

> I remarked that the Z value of the output is stuck at the max value, although the z value in the raw tracker data is changing correctly ie the soft works to detect my z motion. few minutes later I [...]

This is the kind of setting I'm using with ETS2, ATS or MSFS2020, with Z axis inversed checked, works pretty fine:

[image]